apps4av/avare #132: Port project to Android Studio?

postalservice14, 2015-05-13:
@apps4av Are you interested in porting this project over to an Android Studio project, since the Eclipse ADT is no longer supported?

I can create a PR for it if you'd like.

apps4av, 2015-05-14:
Will check the Cap tonight and then merge for next release, unless some
issues found.

Regarding AS, we will have to talk internally about this. We have tried
once before to use AS, and it disappointed us a lot. Maybe its better now.

postalservice14, 2015-05-14:
I really quickly used the AS importer and ran the app from AS after importing, and it seemed to work great.  Not sure what issues you ran into before.

apps4av, 2015-05-14:
We had the AS crash, hang, debugging wrong line, auto fill incorrectly etc.
I hope all these issues are fixed. I suggest playing with it a bit then
reporting back.

postalservice14, 2015-05-20:
I've played around with Avare after porting it over...

I tried debugging, a couple of different virtual devices (and real devices).  It never crashed for me, and was awesome to work with.

I use Android Studio for my other projects and love it.  Haven't had any trouble with it.  So maybe the issues you've had in the past have been resolved.

apps4av, 2015-09-01:
Done

closed 2015-09-01 by apps4av