apps4av/avare #143: Add SIGMETs, AIRMETs, and SUA to ADSB

apps4av, 2015-06-07:
Currently they are ignored from the data.

cyoung, 2015-12-17:
Interested in collaborating on this?

https://github.com/cyoung/stratux/blob/newprotocol/getairmet.go

https://github.com/cyoung/stratux/tree/newprotocol/uatparse

     AIRMET KBOS 282045 BOST WA 282045
     AIRMET TANGO UPDT 3 FOR TURB VALID UNTIL 290300
     AIRMET TURB...ME NH VT MA CT NY LO NJ PA LE WV MD DC DE VA NC
     AND CSTL WTRS
     FROM 60E YQB TO 30NNE ENE TO 30E JFK TO 40ESE ECG TO 60NE ILM TO
     40SW RIC TO 40NW CSN TO 20ESE EWC TO 40S YYZ TO MSS TO YSC TO
     60E YQB
     MOD TURB BTN FL320 AND FL420. CONDS CONTG BYD 03Z THRU 09Z

[image]

     $ cat test-data/example.dump978 | ./getairmet -stdin

      [
        {
          "FISB_month": 7,
          "FISB_day": 28,
          "FISB_hours": 10,
          "FISB_minutes": 21,
          "FISB_seconds": 0,
          "Product_id": 8,
          "Text_data": [
            "NOTAM-D KDTW.07\/518 281021Z !DTW 07\/518 DTW TWY L CLSD 1507281021-1507282300\n",
            ""
          ],
          "Points": {
            "dpsb6": [
              {
                "Lat": 42.233325,
                "Lon": -83.209761,
                "Alt": 0
              }
            ]
          },
          "ReportNumber": 12518,
          "ReportYear": 15,
          "LocationIdentifier": "KDTW",
          "RecordFormat": 8,
          "ReportStart": "07-28 10:21",
          "ReportEnd": "07-28 23:00"
        },
        {
          "FISB_month": 7,
          "FISB_day": 22,
          "FISB_hours": 1,
          "FISB_minutes": 6,
          "FISB_seconds": 0,
          "Product_id": 8,
          "Text_data": [
            "NOTAM-D KHYX.07\/008 241305Z !HYX 07\/008 HYX OBST TOWER LGT (ASR 1006483) 432551.00N0835624.00W (3.4NM W HYX) 802.5FT (207.7FT AGL) OUT OF SERVICE 1507241305-1508241305\n",
            ""
          ],
          "Points": {
            "dpsrn": [
              {
                "Lat": 43.454811,
                "Lon": -83.718828,
                "Alt": 0
              }
            ]
          },
          "ReportNumber": 12008,
          "ReportYear": 15,
          "LocationIdentifier": "KMNN",
[... 1695 lines cut ...]
          "FISB_minutes": 0,
          "FISB_seconds": 0,
          "Product_id": 8,
          "Text_data": [
            "NOTAM-D KAOH.07\/024 251617Z !AOH 07\/024 AOH OBST TOWER LGT (ASR 1225648) 404908.60N0840013.90W (6.8NM NE AOH) 1157.2FT (307.1FT AGL) OUT OF SERVICE 1507251617-1508091617\n",
            "",
            ""
          ],
          "Points": {
            "dphre": [
              {
                "Lat": 40.727421,
                "Lon": -83.884395,
                "Alt": 0
              }
            ]
          },
          "ReportNumber": 12024,
          "ReportYear": 15,
          "LocationIdentifier": "KTOL",
          "RecordFormat": 2,
          "ReportStart": "",
          "ReportEnd": ""
        }
      ]

apps4av, 2015-12-17:
Great! We have the decode but someone need to find time to connect data
with shapes. Its been on low priority because no one has pushed for it on
the forum.

cyoung, 2015-12-17:
Where's the decoding at? avarehelper has

      Logger.Logit("Not implemented product 8");

etc. Can't find anything in the avare source either

apps4av, 2015-12-17:
Sorry I meant I have decode in C. I need to convert it to Java then connect
the information to shapes. But there is no urgency as there is no demand.
SUA makes sense to be done first but long term products like
AIRMETs/SIGMETs are less priority. I agree that they need to be added for
completeness.

cyoung, 2015-12-17:
NOTAMs seem like one of the simpler additions (productid 8) - and useful.

Is that C code anywhere on github? Would be great so as not to rewrite this.

apps4av, 2015-12-17:
You have to request it from an ADSB vendor and we got it from Dual. I am
not sure if we can share it without their permission.
 Yes, NOTMAs will be great. Will keep an eye on your development and as
soon as this becomes priority, will notify. Thanks

JonMcLin, 2015-12-18:
It's 600 (relatively-sparse) lines of commented "go" code, which is pretty readable to an old-school C developer like me, and presumably also a Java developer.

I want to know where to find this document, which is in the comments  (no joy on Google):
Aero_FISB_ProdDef_Rev4.pdf

cyoung, 2015-12-18:
https://github.com/cyoung/stratux/tree/newprotocol/notes

My Go code is pretty rough now - just testing the waters to see if any
interest. Could use gomobile to compile this into a library eventually

mooneyac, 2017-07-07:
"But there is no urgency as there is no demand."

I think adding SUA / TFR from stratux, not just through the internet updater, is VERY important. TFR data is part of FIS-B, and the stratux correctly receives the data, but then avare doesn't display it. I don't want to fly with stale internet data, FIS-B is for LIVE data! So please show it to us!

apps4av, 2017-07-07:
I understand need to fully use ADSB capabilities. TFRs and SUA are issued
well in advance so getting them reliably on the ground is much more useful
than getting them in the air. This bug is about Sigmets which we will add
soon as they are weather products and are updated frequently.

<https://github.com/apps4av/avare/issues/143#issuecomment-313791067>, or mute
the thread

mooneyac, 2017-07-07:
- Perhaps I'm in the wrong issue here, as this is not primarily about TFRs. 
- "issued well in advance so getting them reliably on the ground is much more useful
than getting them in the air" is wrong logic. TFRs change, we just had that in the NY metro area last weekend with multiple TFRs (Morristown, Bedford). a) not all TFRs are issued well in advance, b) even when they are issued well in advance, they can (and do) change any time, c) allowing for the reception of live changes to TFRs was a primary reason to get them into the FIS-B data feed in the first place. If there were no need to get the live changes of the TFRs, then they wouldn't have included them in the FIS-B data feed. Apologies if this is on the wrong issue (SUA, AIRMETS, SIGMETS), I'm new to github, and there doesn't seem to be a dedicated "include TFR data" issue.

cyoung, 2017-07-07:
TFRs are broadcast over FIS-B both as NOTAMs (I think these are also not parsed by Avare) and using the same mechanism as graphical AIRMETs.

cyoung, 2017-07-07:
Attached is a document from MITRE.

Here is an excerpt from the document, speaking to the importance of using FIS-B for TFR data:

>Recently, aviators in the United States have found a considerable number of restrictions placed on airspace around prominent landmarks and facilities. There have been numerous situations when changes to temporary flight restriction (TFR) NOTAMs have caught pilots off guard and the pilot subsequently transgressed an airspace unintentionally. The airspace around Camp David in Thurmont, MD is a good example of an airspace that changes regularly with little notice. FIS-B offers a mechanism to greatly simplify the delivery of TFR NOTAMs and do so in a manner compatible with available navigation display capabilities.

See Section 6 "Graphical Overlay Record":

>The Graphical Overlay Record is a component of the Aerodrome and Airspace FIS-B Payload containing the fields for the graphical depictions of NAS Status information (e.g., TFR NOTAM and SIGMET reports).

Unless they've released a new version of the SDK, the Dual code referenced above is broken for parsing these products, which was the reason I started writing the above referenced Go decoder code.

[image]

[Aero_FISB_ProdDef_Rev4.pdf](https://github.com/apps4av/avare/files/1132388/Aero_FISB_ProdDef_Rev4.pdf)

apps4av, 2017-07-07:
I don't think TFRs with ADSB will save you. It's good to have but its not
as important as your reference suggests. Are stadium TFRs  transmitted on
ADSB too? I am not arguing about not adding it. It's just low on priority
on our side.
If you want my opinion get them on ground then keep in touch with ATC. If
you insist they be prioritized, we will then have an internal discussion.
Thanks

mooneyac, 2017-07-07:
- although most TFRs are issued well in advance, the internet-data solution still has its problems. I now have a workflow problem. It forces me to update from the internet just before I take-off, JUST to get the latest TFR info, as there could have been changes. I now have to know the wifi password of the FBO, some don't have wifi, etc. If I could get all the important info live, I could drop the extra get-internet-data item. Some existing/missing features have a workflow impact!
- stadium-TFRs are not transmitted via ADS-B. They can't, the FAA doesn't have any data infrastructure for them. It's officially the pilot's obligation to know about stadium TFRs.
- of course we can get them on the ground and keep in touch with ATC. I'm not saying there are no ways to deal with it. We could use Garmin Pilot, which covers ALL FIS-B data. But I don't think it's a valid feature request rejection to say "there's ways around it". A feature request rejection should be based on its own inherent merits, not on extraneous arguments.

cyoung, 2017-07-07:
Stadium TFRs aren't transmitted over FIS-B. Those ones are definitely planned ahead of time and the risk of those changing in flight is low. Another missing piece is combining internet data and ADS-B data, I think that was mentioned on the Google Groups forum.

It's up to you how it's prioritized, of course. If you want to work together on it as I offered in Dec 2015 then I can still try to help out with it. Implementing the graphical overlays will bring TFRs, SIGMETs, etc.

apps4av, 2017-07-07:
I think we have to disgree on this till something changes in current
method. You do what you have to do to get them best for you. Since game
TFRs do not get transmitted on ADSB yet, that speaks for itself. One more
question, if you don't get a TFR on ADSB because of interference or data
corruption can you tell that to an inspector as a valid excuse instead of
no information on internet and no information by ATC?

apps4av, 2017-07-07:
Cyoung, definitely. What can we do to get these in. I remember you sending
me email but I need decode code not JSON.

mooneyac, 2017-07-07:
What I do is that I take periodic screenshots. After the flight, I shove them up to Google Drive directly from the tablet, and delete after one year. The screenshot includes a time-stamp. Also good is a ceiling-mounted camera (I use VIRB Elite), that records my whole panel, including the tablets, all the time. It's also the only way to show what's on the panel-mounted GPS map, because you can't take screenshots from that. You can never have enough exculpatory evidence when governmental peeps are after you.

Although I still don't understand why you are raising this point for the feature implementation question/decision. Whether or not interference or data corruption are valid excuses to an inspector, compared to no info from internet or no info from ATC, is irrelevant for the feature inclusion/rejection decision.

I think it's clear it's a desirable feature, just a prioritization issue. Unfortunately I can't participate, these are not the programming languages I normally use.

apps4av, 2017-07-13:
Added all (3), SUA, AIRMETS, SIGMETS for release 7.7.1

closed 2017-07-13 by apps4av