apps4av/avare #146: Draw mode not working on Nexus 5

nickdavies, 2015-07-10:
When I switch between draw and pan mode both pan instead of drawing and draw mode is missing the erase button.

Phone model: Nexus 5
App Version: 6.0.9
Android Version: 4.4.4

[image]
[image]

nickdavies, 2015-07-10:
This could quite easily be user error as I just started using the app. If you need anymore info let me know

avcomware, 2015-07-10:
Hi Nick.I was unable to duplicate your problem. 
You may want to try and see if it does the same thing in Portrait or Landscape, Not Sensor? It works on my devices if I have the Display>Screen orientation set to either Portrait or Landscape or Sensor.Sensor Mode can do some strange things.
If problem continues post on:https://groups.google.com/forum/#!forum/apps4av-forumfor others to see and respond.
You may want to try and see if it does the same thing in Portrait or Landscape, Not Sensor? It works for my devices if I have the Display>Screen orientation set to either Portrait or Landscape or Sensor.Sensor Mode can do some strange things. TV

>>When I switch between draw and pan mode both pan instead of drawing and draw mode is missing the erase button.Phone model: Nexus 5
App Version: 6.0.9
Android Version: 4.4.4<<

apps4av, 2015-09-01:
This was in Track Up. North Up was fine. Fixed.

closed 2015-09-01 by apps4av