apps4av/avare #170: Deleted/superseded charts not removed on update

mark-chromium, 2015-12-22:
I flew into KHTO last week, IFR, using Avare on my tablet for charts. The winds were favoring runway 28. I noticed that Avare offered the following approaches to that runway:

 - RNAV-GPS-RWY-28
 - RNAV-GPS-Y-RWY-28
 - RNAV-GPS-Z-RWY-28

Y, Z, and unsuffixed. That’s suspicious, and it didn’t agree with what was in the on-board GPS’ database. I put Avare down for the rest of the flight.

On the ground, I looked at the charts in Avare more closely:

 - **Y** was _Amdt 1A 20AUG15_, chart date _15232_, from volume NE-2, 10 DEC 2015 to 07 JAN 2016 (_AIRAC 1513_, **current**)
 - **Z** was _Orig-A 20AUG15_, chart date _15232_, from volume NE-2, 10 DEC 2015 to 07 JAN 2016 (_AIRAC 1513_, **current**)
 - **Unsuffixed** was _Orig 07MAY09_, chart date _15008_, from volume NE-2, 05 FEB 2015 to 05 MAR 2015 (_AIRAC 1502_, **expired!**)

The unsuffixed approach is not current. It stopped being published earlier this year, superseded by the Y and Z approaches. **Avare should not have shown it to me.**

I checked my phone, which also has Avare installed, and it correctly shows only the Y and Z approaches. But my phone is only a month old. My tablet’s much older and has had Avare installed for quite some time, including when the time before the Y and Z approaches existed.

I suspect that what happened was that Avare downloaded the expired unsuffixed approach when it was current, but never deleted it when it should have been unpublished and superseded by the Y and Z approaches.

I don’t think I need to explain how serious a data problem this is. **Avare should really be purging all old charts when updating to a new cycle. Under no circumstance should it be presenting expired charts alongside current ones** without any indication that some might be expired. I was lucky to catch the discrepancy, but it could have made me look like a flippin’ idiot…or worse.

gustafson, 2015-12-22:
Hi Mark, 
While I agree we should probably rediscuss the default behavior on
this, this is known and (reasonably) well reported/documented behavior.
 I'm copying this to the forum to solicit user preferences and to again
report/remind of the default behavior. 
For those who don't want to "risk" _anything_ past the expiration date,
you must manually delete the corresponding directories (plates,
minimums, area, afd, etc) before upgrading to new.  (Some have
expressed a desire to not "auto delete" things as they expire. Both
sides of the discussion have made reasonable points.)
A few quick notes: superseeded plates get replaced unless the file name
(based on approach name) is changed.  I know that may not make you more
comfortable, but... it is a small percentage that are left over. Also
when you say "without indication"... well that is a sticky point
because (one might argue that) the valid dates are displayed on each
plate, and should be part of the initial scan when opening any plate. 
 (I agree with your sentiment and support auto-delete personally (to
avoid the potential for confusion); I'm not trying to be difficult
here. But a point is that Avare never alleviates the pilot of their own
responsibility and avare presents the critical necessary info on the
plate.)
Pete

apps4av, 2015-12-22:
Thanks for reporting.
I will add code to delete the older plates. The code existed but was not working due to change in Download products package change a few cycles ago. Look for fix in release 7.1.2.
Regarding the issues you had, I suggest for future, always verify dates imprinted on AFD/Plates. The data is enormous and we make mistakes.

mark-chromium, 2015-12-22:
Thanks. I might have noticed that it was expired if I got to the point of actually briefing the chart. I put Avare down as soon as I recognized I had a data quality problem, and didn’t pick it back up for the rest of the flight (or return trip) until I could figure out what was going on.

gustafson, 2015-12-23:
Gentlemen,

If a person wants to 'keep' an old document how would they know which ones 
NOT to delete?

Solution? -> Add an option for the user to chose if they want
(1) Auto Update
(2) Do it manually
(3) Give the end user an additional option [+Get]  [?Update] [x Delete] [* 
DoNot Delete/Update]
     The * would have the warning that that chart would not be deleted nor 
updated (until the user wanted to select a different option)
      AND put say, a RED band around the outdated chart display when/if it 
was ever displayed with the words in 'reverse screen' within the band that 
the chart was not current.

This provides a solution for BOTH sides of the 'consumer' debate ....and I 
would think protect Avare.

From the position of safety in the aviation system, I ONLY want current 
charts.

ROaks
==================================

gustafson, 2015-12-24:
I've got to admit, as a VFR pilot it's difficult for me to assign more 
weight to avoiding embarrassment (or even legal action) than I assign to my 
safety. And it's very difficult to see how having *no* information is 
better than having info that's a month, or even 5 years, out of date. I 
still keep sectionals in my plane that are so old they are coming apart at 
the folds, but for ultimate backup, they are far better than nothing. Very 
few airports or cities move enough to keep me from finding them using a 10 
year old chart.

Being notified that a chart is out of date when I start the software is 
fine and good, but having a critical part of the chart data disappear when 
I start a trip home is not my idea of fun. Kinda like having a sectional 
printed with time-delay disappearing ink.

FWIW,

Charlie

closed 2016-01-02 by apps4av

hmehendale, 2016-12-24:
I think the problem is that having expired IFR charts is no good, or worse, you're causing conflict with others in the IFR system. It's fine for a VFR pilot to have old charts which is better than nothing, but flying in IMC is a whole different situation - you *have* to be able to rely on information you have, because you don't have any external references.

For instance, FAA has started the process of decommissioning VORs, and therefore any approaches that depend on the VORs to be decommissioned must be decommissioned as well. If I'm using old IFR charts ("procedure plates", technically), I might end up relying on an expired chart when I really shouldn't, causing bad things to happen. It *may* be OK to show the old charts for a few days (or for an extra update cycle), or with a bold EXPIRED overlay on it (it's easy to miss the chart date), but any EFB shouldn't really be showing anything that's expired, that's kind of the *point* of using an EFB. Also, to @gustafson's point about charts being updated unless the filenames change -- if an approach is decommissioned, there will never be a new file. Yes they may have been few and far in between, but a LOT of VOR approaches (several hundred) are to be decommissioned over the next few years. (e.g. KSCK VOR was decommissioned recently, as were both the KACV VOR approaches to 1 and 14)

apps4av, 2016-12-24:
Is that a bug report?
When was the last time you updated and the old charts were not deleted?
Which version are you using of the app?
How do you reproduce it?
Please write bugs here. Philosophy should be on the forum.

apps4av, 2016-12-24:
The bug is closed. It might have been a problem at one point but it was fixed.
If you notice it, give a version number of app, OS, and steps to reproduce and will reopen.