apps4av/avare #173: Background for second row of timers/tools always appears at app start, even if empty

pballer2oo7, 2016-02-09:
I have Background turned on for the timers/tools at the top of the screen. If I disable (set to "None") all of the second row of tools/timers, the background for the second row vanishes, as expected. 

However, when I close and reopen the app, the second row of Background appears, even though the second row is empty. To get it to vanish, I must double tap one of the slots on the second row, and immediately tap "OK" in the "Choose field to display" picker that pops up (still set to "None"). See images below.

Unexpected two rows of background: http://imgur.com/6H16i2Z
Expected single row of background: http://imgur.com/5VpCdr8

pballer2oo7, 2016-07-14:
should we can change the "Show Background" label to something that's a little more intuitive?

apps4av, 2016-07-14:
Yes send a pull request. We call this top area status lines.