apps4av/avare #175: Add 3D view

apps4av, 2016-02-26:
A 3D view of terrain is needed.

annerajb, 2016-02-26:
Where would be get the terrain meshes?? They would also need to be split by zones to reduce download time.

apps4av, 2016-02-26:
Will use elevation charts. The code is ready, but needs some tweaks.

cyoung, 2016-02-26:
http://viewer.nationalmap.gov/launch/

annerajb, 2016-02-26:
Ah ok i saw a issue and Thought it was a feature request didnt knew it was
in progress.

gustafson, 2016-02-26:
Hey Z,

This is awesome. I know you already have at least alpha or beta code
working.

The intent of this is entirely visual.  But I'm concerned that the
resolution will be insufficient both in the planar projection and in the
elevation.  The current shaded relief tiles are strictly for low res
"overview" visual consumption, I think we can do some awesome stuff but
think we should not use current shaded relief as the dataset.

I just pulled some data from our shaded relief... it appears that our
current resolution has about 105 (or more) meters per pixel at best,
(563 meters per pixel in alaska). Whereas the current national datasets
(provided below) are at our choice of 10 meters or 30 meters per grid
(north south, varying but similar in east-west).

http://nationalmap.gov/3DEP/3dep_prodserv.html

And our 8 bit grayscale maps (I think) 24 meters per level resolution.
Not great... most of Michigan is essentially "flat".  I'd shoot for 12
bit gray (1.5 meters resolution).  Or we could use 8 bit but have a
non-linear elevation model, which would allow the lowlanders to have
some visualization.

I think the "cost" of doing different than shaded relief is relatively
low (using tile layering).  We could break it up to be more than just 9
regions to keep download and storage manageable.  (I actually think we
should replace shaded relief altogether, rather than add yet another
dataset.)

I'm happy to mock up some new images for you.

Pete

francopapa, 2016-02-27:
Hi all
as i have noticed, some elevation chart that are availabe from the web are embended with an xml file 
that describe the content from the chart. Question: is possible to use a table (database) that 
contain the elevation data instead of the elevation chart/tiles?

gustafson, 2016-02-29:
I think it is fundamentally the same task... storing a long integer or
floating point value on a regular grid. (That is essentially what a
bitmap image is.) Storing as text files will make the challenging
slightly more daunting, since ascii text does not store numbers as
efficiently as does binary.

One way it would be more efficient is if you could store only the data
required for meaningful changes in elevation.  That is, if you didn't
store redundant data in flat areas (to a certain tolerance, ie one data
point covers a large area).  I'm not sure how much can be gained from
it... the country is big and not particularly flat.

I could take a look at triangulating the data and storing only what is
required... like with a binary stl file?  It wouldn't take me much but I
doubt it will come to much in the end.

Pete

francopapa, 2016-02-29:
Hi Pete
thanks for the reply.

I believe, that if you request more precise elevation data, then we become allways more data to 
store, independently if we use charts (tiles) or text file (database).

I have beginn in the last year to make these charts for europe, but never finished this job. But i 
have found another two source for the tiles in the USA and Europe: if you need a sample, then dont 
esitate to ask me.

Question: live you in europe or usa?

Franco

apps4av, 2016-06-02:
Done. 7.2.7

closed 2016-06-02 by apps4av