apps4av/avare #176: Use Android native multi-touch controls instead of outdated library

r-oleary, 2016-03-01:
The MultiTouchController is no longer needed as Android provides native controls for multi-touch scaling.

apps4av, 2016-03-01:
Is this native control supported in support libraries for older versions and what api are you referring to?

r-oleary, 2016-03-01:
That's a great question. What is Avare's position on backwards compatibility? Is there some documentation I can read through? Similarly, as someone who would like to start contributing to this project, are there any design docs or anything?

This is what I'm referring to: http://developer.android.com/reference/android/view/ScaleGestureDetector.html

Looks like it was added in API level 8.

I actually have it mostly working for LocationView already

apps4av, 2016-03-02:
If you submit a pull request I will take a look.
Avare supports down to OS version 9. We can look into going up if there is
a case for it.
There is no design doc as project was started on a whim. Will help if you
have questions. There is a developers group if you want to discuss there.
Thanks!

r-oleary, 2016-03-07:
See what you think. I did it only for LocationView and PlatesView so far. The others should be easy. It's not perfect - LocationView needs some tuning, but PlatesView is pretty good (the tile reloading on LocationView can cause artifacts). https://github.com/r-oleary/avare/tree/use_native_multitouch

apps4av, 2016-03-08:
The LocationView is where the Touch code needs to be modified. As the zoom
changes, a macro factor changes tiles resolution. It becomes a little more
interesting there and thats why there are issues with new code in
LocationView?

r-oleary, 2016-03-08:
Yeah, it's all kind of complex. What you'll notice with the branch I sent you is that, on LocationView, when you use one multi-touch movement to scale across macro factors and then pan, that the panning doesn't scale accordingly. I can change it to update mScale.updateMacro which will fix the panning, but if you do that, then you have other problems with tile calculations and such. It's just pretty complex.

One thing that I was trying to do for a while was get the tiles to update as you were inside of one multi-touch gesture -- this became pretty complex since loading the tiles takes pretty long. It wasn't very fluid (since one multi-touch gesture will trigger the handler very often -- multiple times per second) and the Async task loading the tiles kept getting canceled by another event coming in. 

Overall it made me think about the whole method of loading the tiles in this way (which is why I asked about design). Other apps appear to do something a bit different that provides a better loading experience. Did you consider using the Google Maps API with overlays to do this instead? I don't know anything about it, but I wonder if it would be easier to leverage all of their loading/panning/scaling logic and just provide new overlays.

Check out the PlatesView though - if you like this approach/interaction, I'll make that change to the other views and figure out the LocationView problem.

Where is the dev group that you mention? I have some other things that I'm thinking about...

apps4av, 2016-03-08:
Go ahead if you want to use google maps and overlays. Will prefer surgical changes though instead of complete overhaul. Drawing engine does have its flaws but it is very fast and battery and memory efficient. We can make a comparison when you are done with LocationView. I will check PlatesView and merge.

r-oleary, 2016-03-08:
Ok, maybe I will try it out.

Don't merge PlatesView yet, as I broke drawing on it when I did the change. Assuming you want that to keep working (I personally don't find it useful), I'll make that work again and send you a PR for just that change.

Another related question - I see the old multitouch controller being used in AfdView as well, but I can't figure out why. It seems like AfdView used to use bitmaps or something? But now it's just a list? So, either I'm missing something or there is some code there that can get cleaned up. Any perspective on this?

Lastly, where is the dev group that you mention?

apps4av, 2016-03-08:
AFD view shows bitmaps if you select them from top drop down menu. You will
need to download AFD for your area first before it suggests anything.

r-oleary, 2016-03-12:
Here's a PR just for PlatesView for now: https://github.com/apps4av/avare/pull/177

r-oleary, 2016-05-31:
@apps4av any update here? Do you not like this feature/PR?

apps4av, 2016-05-31:
The push request is great. However, its not merged yet since LocationView is still using the old library. Hence merge will not close this issue - "Use Android native multi-touch controls instead of outdated library". Thanks.

r-oleary, 2016-05-31:
Ok, makes sense. I'll wrap up location view too. It still has some weird artifacts, but those artifacts are present today as well, so it is no worse.

apps4av, 2016-05-31:
One more thing. Could the new code be put in a separate class and the Views use it with touch events?

r-oleary, 2016-06-05:
PR updated to include all views (including new ones) and some refactored code per @apps4av : https://github.com/apps4av/avare/pull/184

closed 2016-06-05 by r-oleary