apps4av/avare #182: Plate view switches to Airport Diagram randomly

r-oleary, 2016-05-31:
When flying an approach and having Plate view open to the correct plate, the app will sometimes just switch to the Airport Diagram. This can be a real problem if you're in the middle of the approach in IMC and glance down to find some piece of information and find instead that the wrong chart is being shown.

gustafson, 2016-05-31:
This seems likely to be related to the "auto show airport diagrams"
option, which triggers based on ground speed.  If you are a slower
airplane... maybe your approach ground speed is slow enough to trigger?
 Try disabling this option.

r-oleary, 2016-05-31:
I saw that as a feature request - I didn't realize that was already in the app. Regardless, I'm flying a 182 and this just happened yesterday with my ground speed around 90 knots, so that doesn't seem like the culprit. 

Looking at the code, it looks like we look for that speed 1 time. IANAE (yet), but can't it be the case that we get one inaccurate datapoint every once in a while (like a speed of <40 kts)? That would cause a premature change.

I like the idea in the code of using alititude as well as (or instead of) speed to switch the plate.

But ya, for now I can turn that option off until it works better.