apps4av/avare #194: Chart is lagging behind aircraft movement with time

masterhard, 2016-06-27:
This is a new problem introduced in one of last month updates (maybe, last two months).

In the chart mode chart is moving slower than airplane actually flies. However, flight plan route rubber band moves properly thus getting out of sync with chart. Tapping on "center aircraft" button (at the bottom center of the screen) puts everything in the right place but chart starts to drift again after a while.

On a flight from KHWD to KEMT I was observing myself on the chart abeam KSJC while I was abeam KCVH already. This issue reproduced multiple times this weekend on San Francisco and Los Angeles sectional charts.

apps4av, 2016-06-27:
7.3.2 release had this issues and fixed in 7.3.3. It was done within a day.
Update of app will fix it for you.

masterhard, 2016-06-27:
That was a quick response! Thanks!!!

closed 2016-06-27 by apps4av