apps4av/avare #203: While in simulation mode, teleporting to destination from Find view not working

xargsgrep, 2016-06-29:
In simulation mode, selecting a new destination does not teleport to that destination, and similarly center button does not either. Destination is not set at all if you do it from Find. But you can set destination if set with methods in Map screen. Similar issue is expected in Near, Plan, and Find due to intents with Destination sent to the Map activity may not be going to Map fragment.

xargsgrep, 2016-07-02:
previously this would be done in the `ServiceConnection.onServiceConnected` method when the activities would be resumed and the service was bound. the fragments however are only shown and hidden, so their lifecycle methods (eg onPause, onResume) are not called and the `ServiceConnection.onServiceConnected` method isn't called at that point. moved the code out into another method `initializeFromService` and call it when the fragment visibility is changed in `onHiddenChanged` in addition to the `ServiceConnection.onServiceConnected` method.

closed 2016-07-02 by xargsgrep

vinodkutty, 2016-07-04:
Thanks for fixing this (and other bugs recently)! This is one of several issues that I encountered as a new user and continues to frustrate me as an infrequent user (i.e. I haven't fully absorbed various bugs/UI choices that aren't intuitive to me). Much appreciated.