apps4av/avare #207: all fragments remain running in the background when tab is switched

xargsgrep, 2016-07-03:
this is bad for things like the 3d view which is relatively resource intensive. also, all running fragments are onPaused and onResumed when going to another activity and returning (going to preferences for example). this causes toasts/snackbars to pop up from views that you aren't even on.

xargsgrep, 2016-07-10:
decided to allow the fragments to continue running in the background, with the below exceptions. if this causes problems we can revisit. to solve the issue of snackbars popping up from fragments that are hidden, i simply only show them if the fragment `isVisible()`.

1. the map view and 3d view fragments cannot be running at the same time since they are the most intensive. so if you are on the map view and you go to the 3d view, the map fragment will be destroyed. similarly, if you are on the 3d view and go to the map view then the 3d fragment will be destroyed.
2. the tools fragment is always destroyed when navigating away from it

closed 2016-07-10 by xargsgrep

apps4av, 2016-07-10:
Thats a very good idea.
3D and Map should stop each other. Although all the Map memory resides in
StorageService. This is to 1. Speed up switching to Map and seeing the map
there instantly 2. To decrease Bitmap allocations and recycling.