apps4av/avare #210: menu button doesn't show after hiding toolbar/tabbar from preferences

xargsgrep, 2016-08-01:
i went in to prefs and unchecked "hide toolbar", but the toolbar isn't there and now the little menu button is gone too - so i'm stuck

JohnW4SBA, 2016-08-01:
> i went in to prefs and unchecked "hide toolbar", but the toolbar isn't there
> and now the little menu button is gone too - so i'm stuck

This is unusual behavior for the app.  I presume you've tried Exit / 
relaunch of Avare.  It's probably time to next back up anything you have 
saved, or at least go into Downloads and note the charts you have 
(though without the Options button [if by "menu" you mean the square 
button right above "Map" at the bottom-left]), and then Uninstall and 
reinstall Avare on gPlay.  That does wipe out everything you've saved, I 
think except Tracks.

If you want to try avoiding that, you could also try restarting your 
device and ensuring that nothing else is running.

That's all I can think of.  Please let us know what you find.

apps4av, 2016-08-01:
That makes me think that all development related issues should only be visible to guys doing development. But I do not know how to do that on github.

JohnW4SBA, 2016-08-01:
Sorry Ahsan, didn't notice this is on GitHub until after Send when I was 
puzzling over the behavior you described.  I was about to add another 
reply to the apps4av-forum when I finally noticed.  Don't I feel silly! :}

JohnW4SBA, 2016-08-01:
> development related issues should only be visible to guys doing development

I seem to recall that you added me when this latest update collaboration 
began, in case I could help.  I do enjoy skimming the discussion even 
though much of it goes over my head, but feel free to drop me if you 
like.  If you decide to leave me on for later when I might actually be 
able to contribute, I'll *try* not to post before engaging brain in 
future. :)

apps4av, 2016-08-01:
Your comments anywhere and everywhere are most welcome. I was just
suggesting that somehow we should be able to mark some issues as 'out in
the field' versus 'in development' more clearly. The Milestone field in the
issue should help but is not very clear.

xargsgrep, 2016-08-02:
fixed - https://github.com/apps4av/avare/commit/74077fa2c09e003712ef1be7ca5be15573c439dc

closed 2016-08-02 by xargsgrep