apps4av/avare #211: Adding a colon (:) at end of list creates a new line

apps4av, 2016-08-03:
I think I've found a bug in the lists tab.  Attempting to end a line in a colon results in two lines.  For example, entering "Restart procedure:" in the box and hitting add, will result in the following

Restart procedure
:

Adding a space after the colon seems to be a workaround.

pasniak, 2016-10-10:
in **addStep**
`Mixture Rich::Carb Heat On::Ignition Both:: += "test:".replaceAll(DELIM, "--") + DELIM;`
results in
`Mixture Rich::Carb Heat On::Ignition Both::test:::`

Then, in **getStepsArray** 
`mSteps.split(DELIM)` 
results in 
`{..., "test", ":"}`
because ::: is matched on the first occurrence of ::

We cannot change the separator (because it would break users' lists). BTW: putting colon in front of the list item results in an empty HTML line: https://i.imgur.com/a80N4I6.png.

Obviously colon cannot be at the end of lists out in the wild (because it would be split off into a new line).  I think we can safely rule out colon at the beginning of the line.

A) a fix could be in 'addStep(String step)' 
 1) replacing ":" at step[0] with " :" 
 2) replacing ":" at step[step.length-1] with ": " 

B) another fix would be to replace "^:" or ":$" with some escape sequence, ex "%:%",  and do the opposite operation in getStepsArray. 
 see https://i.imgur.com/BScimZ3.png and PR

dgtaylor9, 2020-11-16:
Found that a single quote in a checklist item prevents that item from being highlighted in green.
Workaround: remove single quotes from check list items.

Avare Version: 8.1.7
Android version: 10, One UI version: 2.5, Kernel version: 4.14.117-19479728
Tablet: Samsung Galaxy Tab S6, model SM-T860

[image]

[image]

[image]