apps4av/avare #215: Add waypoint notifications

pasniak, 2016-08-28:
Hi, I am working on my PPL. While flying my attention span is so tiny so I added waypoint notification support ([notif](https://github.com/pasniak/avare/tree/notif) branch) , which works nicely with Android Wear and Pebble. 

5mi before a Destination, a notification is issued, it is repeated every 60s (updated with distance), unless "Next Wpt" action is pressed.

I guess I will add next preference to disable this by default for people who don't want to be bothered.

Let me know what you think! (note: this is my first Android java code)

Pasniak

JohnW4SBA, 2016-08-28:
> I added waypoint notification support

Very cool Pasniak!  Thanks for your contribution. :)

apps4av, 2016-08-29:
Hello
Please add comments to code to indicate what is happening. I read the code and could not figure out what the code does. All constants should be defined in a constant file or class that has meaning for those constants. Good idea to disable it from preferences by default.
Thanks

pasniak, 2016-08-29:
Thanks for reading my code! Willco. 

I tested the flow when driving. I will add more enroute alerts plus a confirmation of "next waypoint" after "Next" (advance) action on the watch. 

I also started work on capability (dialog) to rename GPS waypoints so that notifications could have meaningful name (in front of @). Xxx&Yyy is not enough.

pasniak, 2016-09-13:
I split off Notification code into a separate file, added comments and waypoint rename feature.  Now the preferences.  Do you think I should allow customization of the constants below?

`                        if (!mDestination.getNoMoreNotifications()
                                && mDestination.getDistance() < 5.0
                                && timeBetweenUpdates > 60*1000) {
/// trigger notification here...
`

pasniak, 2016-09-16:
On [notif](https://github.com/pasniak/avare/tree/notif) branch: Added preference code which switches on notifications (they are off by default).

I am looking for suggestions on when to notify on the approaching waypoint. Right now it is
- when plane is X miles from a WPT (now 5mi) - useful approaching airport and announcing - and
- every 60s thereafter (useful for finding visual waypoints)

apps4av, 2016-09-16:
I read the code.

Still cannot figure out what you are doing.

Are you pushing notifications to the notification bar? Is that API(@19)
pragma for because it works on API 19+?

We already have code for notifications, but we do not send them to
notification bar, we just blink the screen. And it depends how far you are
from destination when the blink happens.

See for example LocationActivity.java, line

private void blink()

All logic is done in Plan about distance/time for notify etc..

Thanks

pasniak, 2016-09-16:
IMHO Blinking the screen is a bit nonstandard..., I have never seen Avare do it! How do I enable that?  I can hook up Android notifications there.

By creating "normal" android Notifications (in the bar) my code allows:
1) receiving notifications in Android Wear which forwards notifications by Bluetooth ( to a watch ). This is what I want on my Pebble watch. Note this is for VFR flying in a C152 where use of tablet is for "awareness only". The watch vibrates and shows relevant info ( waypoint name, distance to, time to, heading to, at 12 o'clock etc... ) which I can glance on -- I don't even need to look on screen (we are not supposed too....I am referring to "see and avoid" ) -- but it would work an any Android watch.  I cannot tell you how many times I got fixated on the tablet... With the vibration, I don't even need to look at the tablet or my tiny phone screen.  I am aware I am X miles from my waypoint.
2) trigger actions from the notifications ( you press 3 buttons on the watch to switch to the Next waypoint )
3) hooking up Third Party tools ( I use Automate ) which can get the Notifications.  (I was considering text 2 speech extensions which say aloud location, altitude etc. or sending sms when you get to the airport, vibrate the phone)

As for API(@19) - it is the level required to send notifications.  Let me see what happens if I simulate an older device.

I believe none of proprietary aviation apps has any of the features above.  

Pasniak

gustafson, 2016-09-16:
Notifications is a cool feature... as an owner of an android watch.  If
it can't work less than API19, then it should be one of those (now
several) features which may eventually motivate towards a drop of
support for older devices (or have two parallel versions).

apps4av, 2016-09-16:
> IMHO Blinking the screen is a bit nonstandard..., I have never seen Avare
> do it! How do I enable that?

In Preferences->Instrumentation enable Early Notification and set Early
Timer value.

> I can hook up Android notifications there.

That will be good. Duplicate code to be avoided.

> By creating "normal" android Notifications (in the bar) my code allows:
> 1) receiving notifications in Android Wear which forwards notifications by
> Bluetooth ( to a watch ). This is what I want on my Pebble watch.
OK got it. There should be minimal changes in existing classes. Just a call
to a function in a new class will do. Also, adding code to service should
be avoided as well. However you can add getters to it.

2) trigger actions from the notifications ( you press 3 buttons on the
> watch to switch to the Next waypoint )

OK. That should get code from Plan control buttons enabled via Preferences.

> 3) hooking up Third Party tools ( I use Automate ) which can get the
> Notifications. (I was considering text 2 speech extensions which say aloud
> location, altitude etc. or sending sms when you get to the airport, vibrate
> the phone)

Speech in airplanes is not useful but you can vibrate the watch. Vibrating
the mobile device is not useful as it is mounted on airplane.

> As for API(@19 <https://github.com/19>) - it is the level required to
> send notifications. Let me see what happens if I simulate an older device.

Yes, if putting pragma disables it on older devices, then do so. Otherwise
call is a stub.

Please simplify the code using existing data and limit code mods in main
code.
Thanks

> I believe none of proprietary aviation apps has any of the features above.
>
> Pasniak
>
> <https://github.com/apps4av/avare/issues/215#issuecomment-247657225>, or mute
> the thread

pasniak, 2016-09-26:
I would like to display a nice bitmap as the notification icon.

How do I transcode between mDestType (in Destination) and a Bitmap?

I tried factoring out  _public Bitmap getBitmap(String dbType)_ in SearchAdapter  but I see some waypoints have "Base" type (I would expect "AIRPORT").

Should I code a separate transcoding function ("if-type-then-resourceid") between the strings below and bitmaps?

in Destination:
    public static final String GPS = "GPS";
    public static final String MAPS = "Maps";
    public static final String BASE = "Base";
    public static final String FIX = "Fix";
    public static final String NAVAID = "Navaid";
    public static final String AD = "AIRPORT-DIAGRAM";
    public static final String UDW = "UDW";

apps4av, 2016-09-26:
Adapter pattern will translate your data type to a view type.

See SearchAdapter.class

pasniak, 2016-09-26:
Yeah, I reused **SearchAdapter** but it does not transcode "Base" (I get question mark).  Should I add "Base" (Destination.BASE) to the if-else list there?

apps4av, 2016-09-26:
BASE is a destination type, not a database type.  AIRPORT is the correct
database type.
Try using database type from Destination instance.

pasniak, 2016-10-01:
Cool, I got it to work. From the default map coords, when I route (using [->D] button) to IXD, db type of IXD is "AIRPORT" but, when I route to OJC, OJC's type is empty string... I suspect it is related to Destination.mFound == false.  What is the meaning of mFound?

apps4av, 2016-10-01:
Correct.

pasniak, 2016-10-03:
Added ability to name waypoints at long press with +Plan (see http://i.imgur.com/40pTZkE.jpg) and renaming them.  Code around this feature is complicated due to non-object oriented way the names are handled together with coordinates ( in "name@lat&lon") and code "everywhere" expecting "&" to indicate GPS coordinates (this is very hacky IMHO...). 

Do you want me to merge all these commits (https://github.com/apps4av/avare/compare/master...pasniak:notif) for a PR into avare?

apps4av, 2016-10-03:
> Do you want me to merge all these commits (master...pasniak:notif

You should have one feature per PR. Thanks.

apps4av, 2016-10-03:
Please do not put everything in one branch. To get it merged to main, give
small changes related to one feature at a time. The
https://github.com/apps4av/avare/compare/master...pasniak:notif is not
merge-able at the moment.
Send a PR for notification that has just notification code.