apps4av/avare #225: Show Performance for fields without METAR

pasniak, 2016-10-05:
for example MU26 has no X-wind; we could pull METAR from the closest available METAR (KOJC) and calc Performance

apps4av, 2016-10-05:
Good idea. Lets do it. A note, please do it in a manner that changes are placed in separate class with minimal changes to existing code.

gustafson, 2016-10-05:
Or interpolate nearby airports rather than using just the closest? Just
a thought not a big deal.

More important I think that we should color code, or make a note next to
it that it is not based on on-field measurement.

Great ideas!

pasniak, 2016-10-07:
Here's demo of the implementation for my field which has no METAR: http://i.imgur.com/VvEg5GU.png Destroy!  Need to add similar code for TAFs.

apps4av, 2016-10-07:
Nice. The performance should say at top calculated from winds at (other
airport).

gustafson, 2016-10-07:
Very cool... can you implement a direction indication?.  4Nm SW

pasniak, 2016-10-08:
> can you implement a direction indication?
https://i.imgur.com/IycXc22.png 

Does that look okay?

pasniak, 2016-10-09:
I added changes to have a consistent look: https://i.imgur.com/0k3jFC0.png

pasniak, 2016-10-09:
PR #232

gustafson, 2016-10-10:
I think it looks better... but I was a bit confused. 

Is the source metar SW of me? SW of the destination? or am I SW of it? 
It took me a more than a few seconds to decipher it and I'm not 100%
sure I was right in my interpretation.

Since each indicator could conceivably come from some different place, I
think it would be more clear if you did something like the following:
(where each explicitly states the most relevant relevant reference point)

Current Position: 52 nm SW of destination MIO

METAR: (@KGMJ 20nm SE of destination MIO)

TAF: (@KXXX 34nm W of destination MIO)

Performance: (@MIO extrapolated from METAR @KGMJ)

Winds Aloft: (@SGF 57nm NE of Current Position)

Its going to be great!  Thanks for your work.

pasniak, 2016-11-10:
I made the "closest METAR" search circle smaller (now it's 5 miles; was 50) and made it configurable in Weather preferences. #247 
https://i.imgur.com/TczZo93.png