apps4av/avare #226: Improve performance of findClosestAirports query

pasniak, 2016-10-05:
We are running this query on every map refresh; on my PC and the emulator it takes ~0.18s; it can be speed up ~0.07s by adding indexes. Cost of this increase is main.db size increase from 14,290,944 bytes to 15,184,896 bytes.

On my PC, tested using SQLiteStudio:

SELECT *
  FROM airports,
       airportrunways
 WHERE airports.Type == 'AIRPORT' AND 
       airports.LocationID = airportrunways.LocationID AND 
       CAST (airportrunways.Length AS INTEGER) >= 0
 ORDER BY ( (-90.89647063424971 - airports.ARPLongitude) * (-90.89647063424971 - airports.ARPLongitude) + (41.77581680036067 - airports.ARPLatitude) * (41.77581680036067 - airports.ARPLatitude) ) ASC,
          CAST (airportrunways.Length AS INTEGER) DESC
 LIMIT 40;

[08:32:02] Query finished in 0.180 second(s).

CREATE INDEX airports_loc_idx ON airports (
    LocationID
);
CREATE INDEX airportrunways_idx ON airportrunways (
    LocationID
);

[08:33:42] Query finished in 0.110 second(s).

apps4av, 2016-10-05:
Thanks for reporting. Must be a bug creep in. Will check and report back.

pasniak, 2016-10-05:
this query is probably run to show "Near" airports as the page is refreshed live when the plane moves.

apps4av, 2016-10-05:
A very good find. The query can be improved but moving mArea.updateLocation out to NearestActivity will fix the issue without changing any database and its queries.

apps4av, 2016-10-05:
Great fine once again. Fix committed. Database change can be for a major release. Will leave this open till further comments from pasniak.