apps4av/avare #227: Quickly download weather

pasniak, 2016-10-08:
Story: Typically, I have everything "ready to go" (plan) at home, 1800WXBRIEF done. At the airport I want just to get latest weather.  I need few confusing clicks to do that and this get old quickly.  

Solutions: 
1) Maybe a long click on Download would trigger only Weather zip file download? This seems simple to code. 
2) On the other hand, full Weather download costs 404kb (http://i.imgur.com/VvEg5GU.png + wireless internet is v slow at my airport...) so maybe add Refresh Metars button?  My plan would be to (a) replace the (useless...) Navigate/Simulation button (b) hit www.aviationweather.gov/ web service (https://aviationweather.gov/adds/dataserver/metars/MetarExamples.php) to get metars for the list of local stations on the map in CSV.  Sample costs is 3kB for 3 airports: https://www.aviationweather.gov/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=csv&stationString=KWRI+KTTN+KABE&hoursBeforeNow=1&mostRecentForEachStation=constraint

Let me know what you think.

apps4av, 2016-10-08:
I need few confusing clicks to do that and this get old quickly.

+Before you implement these you need to know our policy about data. We do
not upload or download any data automatically. This has been through user
comments and preferences gathered over many years.

> Solutions:
> 1) Maybe a long click on Download would trigger only Weather zip file
download?

+Maybe add a tip that in order to see new weather you need to download
weather. Could add a button that will update weather but there are only 2
weather files. Is that too much work considering it takes 10 seconds to
select and press download when you are on ground? Maybe a better way is to
make download a background process so user can move on to other thing if
data is slow.

This seems simple to code.
> 2) On the other hand, full Weather download costs 404kb (
http://i.imgur.com/VvEg5GU.png + wireless internet is v slow at my
airport...) so maybe add Refresh Metars button?

+Confusing. Which metar? Destination, current area on in the way, or all?

My plan would be to (a) replace the (useless...) Navigate/Simulation button

+The Navigate Simulate button is the basis of app. Removing it is going to
totally change the app usability.

(b) hit www.aviationweather.gov/ web service (
https://aviationweather.gov/adds/dataserver/metars/MetarExamples.php) to
get metars for the list of local stations on the map in CSV. Sample costs
is 3kB for 3 airports:
https://www.aviationweather.gov/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=csv&stationString=KWRI+KTTN+KABE&hoursBeforeNow=1&mostRecentForEachStation=constraint

+How to decide which airports to update? How will the user track which
metar are updated and which are not? It's best to update all or nothing for
UI consistency and user confidence.

> Let me know what you think.

+These ideas at first they seem awesome but we have experimented with those
before and they have not worked well.

Thanks

pasniak, 2016-10-08:
okay, I like control of data downloading stuff as I typically go around without wireless Data.  How about [Refresh] button next to METAR on the Airport page? That will clearly be this one airport and will cost 1kB.

pasniak, 2016-10-08:
I read Help. So "Simulation" is "Browse" mode? [I lack words to describe it... another one would be "GPS-independent" mode] I see it can be useful.  But "Simulation" made me think about some developer mode where position is simulated and I never touched it.

JohnW4SBA, 2016-10-08:
> I like control of data downloading stuff

Me too, though it's more an issue of no data on x/c in flight or at 
dinky airports.  What I've done so far is select Get for just the items 
I want, typically TFRs in my case (obviously not radar if data is an 
issue).  I'm VFR and usually cruising low/slow so weather data is 
continually updated off-screen. :)

> How about [Refresh] button next to METAR on the Airport page?

A more time-consuming coding approach that could meet both our needs 
would be to add a button atop the Find screen and one at the end of each 
line.  So if I've selected half a dozen airports as waypoints on an x/c 
or while creating a Plan, they'd be in the list.  Our home airports 
would presumably almost always be on the list unless it's a really long 
x/c.  The line end buttons could pop up an option box to select the 
specific Download item desired (options ideally retained even after the 
airport scrolls off the end of the list); and the single button at the 
top of course would be Download Selected Updates.  Some day maybe there 
could be an option to also update all TFRs.

pasniak, 2016-10-08:
>  Is that too much work considering it takes 10 seconds to select and press download when you are on ground

yes, if you do that 10 times waiting for weather to clear/wind to drop and are as impatient as me :)  I fly VFR so I am weather addict!

>download?

by "Download" I mean Download button on map screen which leads to chart_download.xml  Yet another way is to add new state to chart_download.xml, say "Quick download" and trigger "Quick download" for the selected items on Download long press.  Anyone can configure what he/she want to download and get it in one click. Still it is 400kB so useful on fast connection only.

JohnW4SBA, 2016-10-08:
> So "Simulation" is "Browse" mode?

I agree another word would be more intuitive, but like you don't know 
what it would be.  Pan is already taken.  It needs to be a short word to 
fit on phone screens.  Non-GPS would be no more clear for newbies, and 
of course any change will annoy some established users.

> I see it can be useful.

I use it often, but rarely in flight since it's easy to zoom/drag to 
look at something and then tap Center when done.  A few times in flight 
it's quicker to use Sim/Find/Airport (like when diverting to a known 
airport), though I then need to re-select Next (I never seem to use Plan 
in flight) and of course hit Nav & Center.

pasniak, 2016-10-08:
@JohnW4SBA re no data: I recently tested METARS via SMS while camping out without roaming and it worked great:
https://i.imgur.com/gjgAVt1.png
https://www.reddit.com/r/flying/comments/2wothc/phone_sms_mms_metars_for_when_your_hangar_wifi/

We could hook it up. (enabled in Preferences)

JohnW4SBA, 2016-10-08:
> @JohnW4SBA re no data: I recently tested METARS via SMS

Interesting idea.  I've usually been able to get a full ATIS/AWOS at or 
near airports via Find and then A/FD to get the number.  Only done it a 
few times though, being less of a meteo-phile. ;)