apps4av/avare #231: Rename Simulation in Navigate/Simulation button

pasniak, 2016-10-09:
(branched from #227). The name "Simulation" for this mode is not clear.  It has "development" flavor. What it does is disconnects on-screen location updates (moving map) from the GPS:
`// In sim mode, set location to destination or last known location`
`// In navigate mode, leave location to GPS location, or last known location`

It solves annoying problem (for example Osmand has it) of the map moving while the user is trying to look ahead.  It also keeps GPS running and avoids reacquiring GPS lock. 

In Google Maps once the user zooms out of current position, "Re-Center" button appears; one click on it sets back Navigate mode.  I personally like this behavior as it requires 1 press to enter the mode and 1 press to exit (Avare: 2 presses for both).  And the button is not cluttering GUI until it is needed.

@JohnW4SBA:
>I agree another word would be more intuitive, but like you don't know what it would be. Pan is already taken. It needs to be a short word to fit on phone screens. Non-GPS would be no more clear for newbies, and of course any change will annoy some established users.

Current Ideas:
> "GPS-independent" mode or "Lock position" 

"Pause" [as in "Pause Navigation"]
"Pause scroll"

JohnW4SBA, 2016-10-09:
> "Pause" [as in "Pause Navigation"]
> "Pause scroll"

Too long imo.  Please try to imagine that button on a phone screen.  If 
big enough font to be readable, it would take full width in portrait mode.

How about buttons with a GPS icon of some sort (globe? satellite? tiny 
version of our current Center symbol?) that would be color for the 
Active GPS follow and Passive for retention of GPS lock with current Sim 
behavior?  Next to the icon would be Act/Pass as abbreviations of Active 
and Passive GPS.  This might be slightly more intuitive than Nav/Sim for 
newbies and pique their curiosity enough to read the Help info (and tip) 
about it.  Once used, it's intuitive and the text label refreshes memory 
for infrequent users.

How about +Follow / -Follow?  The word is short and perhaps more intuitive.

JohnW4SBA, 2016-10-09:
> "Pause" [as in "Pause Navigation"]
> "Pause scroll"

Everyone has used a DVR.  How about having the button use the standard 
Play/Pause icons?  The text could then be Resume/Pause;  Follow/Pause; 
Active/Pause;  GoGPS/PauseGPS (leaving out the space both shortens the 
text and points up the unique nature of the function); or some other 
short word to toggle from Pause.

pasniak, 2016-10-09:
As much as I like play/pause icons they are already used for the plan button activate. But right now the text is "Navigate" and proposed "Pause" is shorter?

Imho "Follow/Pause" is good. Better than "Simulate" ;)

I described functionality to my non-flying friend and she mentioned Pause too...

pballer2oo7, 2017-10-10:
what about **GPS On** and **GPS Standby**?