apps4av/avare #236: Add link to liveatc.net archive

pasniak, 2016-10-10:
I find it great to listen to the tower at the destination airport while driving.  It is also nice for review of my ATC work. I have a wget script on my PC to do so but having to transfer files is pain.

Example:
{airport}-Twr-{date} link would download and open the last valid mp3 archive
http://archive-server.liveatc.net/klns/KLNS-Twr-Oct-07-2016-2030Z.mp3

It would be nice to tag mp3 with local times, apt name or even correlate with recent flight track times.

JohnW4SBA, 2016-10-11:
> I find it great to listen to the tower at the destination airport while driving.

Interesting idea, and I'm in awe of how many ideas you bring. :)

At the same time I notice feeling some concern about feature bloat.  The 
thing I have most liked about Avare is simplicity.  My panel gns430 will 
sit up and bark for skilled operators, but to me .

What I'd most like to see is for Avare to remain small, relatively free 
of bugs, simple and fast.  Then all the niche features could be add-on 
apps.  Is something like that possible?  Would more hooks into the app 
be needed?

pasniak, 2016-10-11:
yeah, John, you are right - we want to keep original codebase and core function safe - right place for the thing like this idea is a **plugin**. (at my work plugins are bread/butter...).  Was something like this discussed?  OsmAnd has them:
https://play.google.com/store/apps/details?id=net.osmand.parkingPlugin&hl=en
https://github.com/osmandapp/Osmand/tree/master/plugins

JohnW4SBA, 2016-10-11:
> **plugin**. ... Was something like this discussed?

I think that was one of the ideas with the "Avare External I/O Plugin" 
and "Bluetooth GPS For Avare" apps.  I don't know if more has been done 
since those were developed, to enable easy creation of plugin/add-on apps.

Seems to me enhanced plugin/add-on capability could be a great focus for 
some work, but of course I have little notion of what that would entail. 
  Maybe first creating a framework for what I/O points could conceivably 
be useful, coding and commenting them in Avare, then documenting the 
process of integrating plugin/add-on apps, and some sort of alpha/beta 
testing program in collaboration with the core Apps4Av dev team?  Quite 
a lot of work I'm guessing, though some could be done by anyone 
interested in creating plugin/add-on apps.

gustafson, 2016-10-12:
Hey.

Forgive me, but I don't see a clear relation of this link to avare? 
What do you plan to do with the mp3 in the app?  Especially since this
isn't a live stream, rather it is something you are downloading after
the fact?

It seems perhaps that the link could be included as a comment in the
flight track text (which can be emailed)?  Is that what you were thinking?

I agree with John though you are an idea generator.  Great!

pasniak, 2016-10-12:
The key is to locate the correct LiveATC archive http link which has the apt code and zulu time : either the latest time -- to listen up on the "situation" --  just hand off the link to the default browser, BTW: after I installed ServeStream I was able to get live streams on my phone.  This is the easiest from the dev perspective (bar configuring the names in LiveATC).

I have hard time to locate my recordings... If you did not write the exact time over the airport you have to listen to bunch of recordings.  The one related to the track indeed could be printed into the track comment.  Note the tracks never worked for me: either I could not find them or the one I found were cut off.

closed 2017-01-19 by apps4av