apps4av/avare #245: Display airport diagrams North up

pasniak, 2016-10-30:
Certain airports' Airport Diagram is rotated 90deg.  Ex: KACY, KLNS

[image]

gustafson, 2016-11-01:
> screenshot_2016-10-30-17-32-34

I don't think there is much we should do about this.  Apps4AV is
distributing FAA generated materials, and is hesitant alter their chosen
presentation. Rather, we do our best to faithfully reproduce the
materials they provide.  I can see arguments on both sides, so feel free
to try to convince us that we are taking the wrong stance here.

pasniak, 2016-11-01:
A rotate button would not change FAA representation any more than looking at the tablet in landscape mode ;-)

cyoung, 2016-11-01:
@gustafson - Isn't there a bunch of processing/interpretation done on the "FAA generated materials", which are then processed into png files? https://github.com/apps4av/avare/tree/master/extra/aptdiags

pasniak, 2016-11-01:
It's tough cookie to locate the north arrow (or the variance angle), isn't it? so maybe a manual rotate "North Up"-style (and keeping the setting somewhere) would just work  [my first intuition was "...so where is "Rotate" button here...?"

apps4av, 2016-11-01:
You cannot add a rotate button as some airport diagrams have rotation !=
90. However rotation can be detected in code by calculating displacement
with 0.1 degree change in latitude and simple trigonometry.

JohnW4SBA, 2016-11-01:
> looking at the tablet in landscape mode

I don't rotate tablet/phone in flight.  Is rotation common for many 
users?  I thought nearly all would mount it somehow to prevent it 
falling or becoming a projectile in turbulence.  I initially imagined a 
rotate mount (built one in fact), but don't want even the momentary 
distraction inside the cockpit.  I guess you're imagining a rotate 
button that would be quick, and would change only the Plate orientation 
for the current viewing of that Plate?  Is there any concern about 
adding complexity, clutter, newbie user confusion, and potential for bugs?

apps4av, 2016-11-01:
Agree with Pete that we won't rotate it in charts generation. Some people
like the way it is published in book.

pasniak, 2016-11-01:
great, I see some consensus. So you think the button would be an improvement?  I don't want to code a button [once I find some spare time...] if other ideas are better.

JohnW4SBA, 2016-11-01:
> some airport diagrams have rotation != 90.

This discussion got me thinking about my use case.  I look at airport 
diagrams in two situations:
1. Out of the aircraft during flight planning to find landmarks (FBO, 
etc.).
2. After rollout to follow taxi instructions.

I won't use a Rotate feature in either situation.  1. It's trivial to 
simply rotate the tablet/phone outside the aircraft, and I've never even 
done that.  2. After rollout is already a high-stimulation moment with 
ATC and other aircraft traffic adding to post-flight reconfiguration, so 
Rotate would just be another distraction.

If #2 is the use case, maybe another strategy would be helpful.  What if 
there were two changes to the Plate screen for rollout:
A. Automatic zoom to margins when Avare auto-switches to Plate, and a 
Zoom toggle button I can tap when off the runway that centers the 
aircraft icon showing only the nearby taxiways.  That helps me orient 
during rollout if there's time for a glance, and takes but a moment to 
see where ATC wants me next.  The zoom could even have a Pref setting 
for auto-zoom when ground speed is below 10kts, that I could then tap to 
toggle if wanting to look for that FBO etc. relative to my location.
B. Track option that always shows my track on the Plate screen for 
airport diagrams.  That would instantly help me orient regardless of 
diagram orientation.  I don't care at all which way is North, when I'm 
trying to get clear of the runway or figure out where something is 
relative to my position.

pasniak, 2016-11-01:
=> 1) cmon, rotating table is lame :-)  

My use case was (on the ground) : "let me see where the taxiways so I can look them up fast when ATC routes me"  once I've seen it's kind of sideways and it would be harder to read I went beck to my paper printout [luckily once I said I was a student they just gave me progressive...]

jackwilliard, 2020-12-08:
Trying to capture the consensus here:
(1) don't modify what the FAA gives us
(2) people still want to be able to "north up' charts without physically rotating the tablet
Proposed compromise: provide a button that allows chart to be rotated if desired.
I only found charts rotated by 90degs, but please let me know if there are other odd ball charts out there. If so, I will try to improve the rotate button.
See check-in above.

n251ea, 2020-12-08:
issue #377 and #417 are duplicates of this issue.

I personally think a simple rotate 90 deg button would be the best way to go, it preserves the ability for a user to simply rotate the tablet if they prefer or rotate just the image.  A quick look showed that any plate that needs to be rotated would be rotated clockwise 90 degrees and are usually the visual depictions for the SID or STAR.