apps4av/avare #246: Avare Take-Off minimums documents confusing PAxx with KAxx airports

hmehendale, 2016-11-02:
I noticed this when I was looking at the information about KAUN (Auburn, CA). It shows the proper approach plates (RNAV GPS Rwy 07), and the alternate minimums document (from the SW2ALT-0 pack), but it shows takeoff minimums from the AKTO-20 pack for Unalakleet, AK (whose ICAO airport code happens to be PAUN). 

I cross-checked this for KACV (Arcata-Eureka, CA), and the exact same behaviour manifests - I see approach plates for the runways at KACV, alternate mins. for KACV (SW2ALT-0), the proper DPs, but takeoff mins for PACV (the AKTO-5 pack).

I also checked (arbitrarily) against North Carolina airports by installing the NC Approach plates/dTPP pack, and sure enough I see the same thing happen there too, for KAVL (shows T/O mins for PAVL), and also KAKH/PAKH.

hmehendale, 2016-11-02:
These are most (all?) the airports in CONUS that overlap with the corresponding PAxx airport code:

KABR KACK KACV KADU KAEG KAGS KAHN KAIK KAJZ KAKH KALB KANW KAOC KAOH KAOO KAPG KAPH KAPN KASH KASL KASN KAST KASW KASX KATA KATW KAUM KAUN KAUO KAVA KAVC KAVL KAWG KAWM KABE.

(I've not checked if the same thing happens with KHTO (New york)/PHTO (Hawaii); the other 13 airports in Hawaii don't have a corresponding valid KHxx airport in CONUS)

avcomware, 2016-11-02:
Hi.
Try turning Off Predictive text, in your Settings? TV

hmehendale, 2016-11-02:
This seems to be happening in `storage/DataBaseHelper.java : findMinimums(String airportId)`, which searches for either `Pxxx` or `Kxxx` codes, and always uses the first result, because it only expects one. But I suspect the underlying problem may be that the airport codes are all passed around as 3 alpha-numeric, instead of `K[A-Z]{3}`, `P[AFHPO][A-Z]{2}` (for Alaska and Hawaii), or `[A-Z0-9]{3}` with at least one number. The vast majority of them aren't going to conflict.

hmehendale, 2016-11-02:
[image]

hmehendale, 2016-11-02:
@avcomware Not sure what you mean. Predictive text in Android settings? What does that have anything to do with this?

avcomware, 2016-11-02:
Hi.
It's possible that the actual choice is chosen for you.
Try it and let us know the result. TV

apps4av, 2016-11-03:
Thanks for reporting. There is issue in faa data. It's not really an issue
but they decided to go ICAO on minimums while everything else is Domestic.

hmehendale, 2016-11-03:
Oh, that explains it. So the check in findMinimums() needs to first convert the 3-letter identifier (e.g. `UNK` or `AUN` to the corresponding ICAO code (`PAUN` or `KAUN` respectively)