apps4av/avare #248: Help on long-press is missing info

pasniak, 2016-11-10:
Here's the text:

> <p>•<b>Long-press</b> any point on the chart (with GPS
>   acquired) to display text at the top of the screen giving
>   approximate compass direction and distance<b>From</b>,
>   followed by exact GPS bearing<b>To</b>, that point in relation to
>   your GPS location. Very handy when contacting ATC or making CTAF
>   calls!</p>

That is somewhat true but really long press will give you the point/apt/tfr etc. info.   (I was looking for a place to add somewhere a doc on airport METAR/Performance display.)  Maybe we should merge the above with the following doc?

>  <p style="padding-left: 30px;"><b>Note:</b> The current <b>Destination</b> used
>   by Avare for various features can be set after a long-press on the chart. Just tap either
>   the button that pops out containing the name or coordinates of the location you pressed;
>   or the button that pops out containing the word "Plan" to send that Destination to the flight plan.
>   <a name="0.1_Find"></a>

pasniak, 2016-11-10:
I propose to merge the two with this text:

<b>Long-press</b> any point on the chart (with GPS acquired) to display a page containing:
- <l> Action buttons row: 
 **->D** sets up the point as Destination,
 **+Plan** adds or inserts the point to the Plan,
 **Plate** displays plates, if the point is an airport
 **A/FD** displays a synthetic Chart Supplement page (click on the top dropdown to go to scanned Chart Supplement)
 **X** to close the page
- <l>either <b>FAA code</b> of the airport or <b>GPS coordinates</b> of the point (in _name@lat&lon_ format, name is user's name for the point),
- <l>Position distance and compass direction and GPS bearing <b>From</b> the point,
  followed by exact GPS bearing <b>To</b> the point in relation to
  your GPS location. Useful when contacting ATC or making CTAF calls.
If the point is an aiport:
- <l>the airport's <b>METAR</b> or the closest METAR found in a configurable radius (in Weather Preferences),
- <l><b>Performance</b> data for use at the airport: Density Altitude, the best runway, and its head/cross winds calculated using winds in the above METAR,
- <l>the airport's <b>TAF</b>,
- <l>the <b>Winds Aloft</b> forecast reported by the closest available meteorological station,
- <l>nearby (?) <b>Special Use Airspace</b> (SUA),
- <l>location of the point with respect to the 4 closest <b>NavAids</b> (VORs),
- <l>data source for the weather and SUA, data downloaded from internet or data retrieved with ADSB,
- <l>data timestamp.
</p>

JohnW4SBA, 2016-11-10:
> I propose to merge the two with this text:

Thanks for doing this.  The website is woefully outdated, so if you do 
this and hopefully other improvements to the in-app Help file I can 
further reduce the website Intro and update the screencaps there.  I'm 
currently planning a trip to HI, NZ & AU including some GA air time, so 
won't be able to do anything other than crucial changes to the website 
until mid-Jan. 2017.

pasniak, 2016-11-11:
The new Help HTML is now in #247