apps4av/avare #249: Winds Aloft in long-pressed too long/busy

pasniak, 2016-11-12:
Here's how WX-BRIEF presents en-route winds: https://i.imgur.com/FmGwmAV.png

1. we could add a similar Preference ("Show Winds Aloft up to" or add "Ceiling" to aircraft data) to shorten the list
2. adopt a similar 6+3 characters table (instead of a formatted string). 
3. Distinguish winds from heading by color (kn in greed or yellow if change per altitude differential>25kn to warn about WS). Distinguish temps by showing below 0C in blue
4. We could also add a Preference "Interpolate Winds aloft" with values VFR or IFR altitudes and accordingly also display interpolated values (up to the ceiling) - typical altitudes for planning.

pasniak, 2016-11-20:
added a preference to shorten the Winds Aloft list: https://i.imgur.com/E1E64uY.png
in https://github.com/pasniak/avare/commit/024cfe40fa5506333e63bc696db9f4f953ff4a3f

apps4av, 2016-11-21:
> in pasniak@024cfe4

How is it displayed? In long press view?

pasniak, 2016-11-21:
Yes, in the long pressed, just where Winds Aloft are right now.  
I worked today on the new HTML table presentation  (this one is cut off at 012):
https://i.imgur.com/cjh7yNq.png
What do you think?

pasniak, 2016-11-21:
I think "light an variable" winds averaging (for planning) could be improved.  For those, direction is parsed to 0, which is fine for display but makes the interpolated direction a bit unrealistic.
For example: wind1 9900 and wind2 2405 :  Aware averaged direction* in between the altitudes could be something like 18deg ((240-0)*fac + 0) % 360 when, in reality it, I think it would be closer to 240deg.  Wouldn't it be better to set direction to the direction of the "layer with value" (here 240deg) if one of the layers is light and variable?
*) in WindsAloft.getWindAtAltitude

pasniak, 2016-11-21:
Another "strange" thing I see is a break point in plan.simulate() is hit every few milliseconds.

  Doesn't that indicate that when a user opens PlanInterface she would constantly run run these (unnecessary because altitude is constant in Simulation - I see it's the plan altitude) re-parsings and re-calculations?
`
        if(mPref.isSimulationMode()) {
            double ws = 0;
            double wd = 0;
            if(mWinds != null) {
                double winds[] = mWinds.getWindAtAltitude(params.getAltitude());
                ws = winds[0];
                wd = winds[1];
            }
`

apps4av, 2016-11-21:
It should be hit every second when GPS updates. Will check.

pasniak, 2016-11-21:
The simulated numbers will only change if the setup (plan, winds, alt) changes; no need to calc every second?

Related question: if WA are used in simulation mode (that is my understanding) only how can the Plan in-flight indicate correct times/fuel? 
Say we fly a round-trip A to B (100nm), 0deg course 100kn with 180deg wind 100kn: on the way A to B the Plan uses current GS (200kn so .5h) but what about GS on the leg B to A?  Do you use some implied real-time crosswind?

apps4av, 2016-11-21:
>
> The simulated numbers will only change if the setup (plan, winds, alt)
changes; no need to calc every second?

That's correct but easier to update everything then watch for change.

> Related question: if WA are used in simulation mode (that is my
understanding) only how can the Plan in-flight indicate correct times/fuel?

It does not. Think of it as a piece of paper. It won't change. Could be
improved.

> Say we fly a round-trip A to B (100nm), 0deg course 100kn with 180deg
wind 100kn: on the way A to B the Plan uses current GS (200kn so .5h) but
what about GS on the leg B to A? Do you use some implied real-time
crosswind?

It only uses wind in simulation not in actual flight.

pasniak, 2016-11-21:
Argh, another "hidden" meaning of simulated :)  How about we drop the 'if(mPref.isSimulationMode())' and use WA to calc "projected" numbers.  Based on google that is what FF does. Maybe also make WA expire on a different (its own i.e. NOAA's) schedule?

apps4av, 2016-11-21:
Yes it's good to calculate winds based plan for real time assuming that a
ground speed of 0 will not make eta infinite.
No let's keep all expiry the same. It's easier to remember to update
everything than to read when every little product expires.

pasniak, 2016-11-23:
WA HTML formatting commit at
https://github.com/pasniak/avare/commit/a466d916ddcc07f5e0b808793e42bfc2f9662578

apps4av, 2016-11-24:
> pasniak@a466d91

Could you submit a pull request? Also is it possible to keep the helper in
utils for now? Will move it to weather package after the merge.
Thanks

pasniak, 2016-11-27:
> Also is it possible to keep the helper in utils for now?

Np. The helper is still in _utils_ but I only _split of_f the WA HTML table logic (into new file utils/WindsAloftHelper.java), just so (a) I don't interfere with any other formatting (which stayed in utils/WeatherHelper.java) - and (b) it is clear the new code is not easily mergable - I spent quite some time refactoring that stuff). You really want me to put it back in...?

pasniak, 2017-02-13:
I have a "new" PR in draft stage with Unit Tests (!) and 100% coverage of my HTML WA table generator which caused some core dumps some time ago.  I am still going to meditate over this code before requesting PR but do you have more ideas for tests?
https://github.com/apps4av/avare/compare/master...pasniak:winds_aloft?expand=1

apps4av, 2017-02-13:
> pasniak:winds_aloft?expand=1

Great!
PR when ready.

pasniak, 2017-02-17:
#270