apps4av/avare #253: onDraw() called too often causing Power drain

pasniak, 2016-11-22:
I saw CPU monitor kernel use fluctuating ~10%.  

On a whim I put bpt in onDraw and I got there many times (nothing changes, I am not moving, I am in Sim mode); the cycles in 10% are from drawMet, displayFieldValue

It's kind of hard to locate why onDraw is invalidated again and again. I put bpt in many invalidate() but they are not called.  Maybe this guy in NavComments.draw() causes invalidation?

`
            // Draw them with background shadow
	        paint.setColor(Color.WHITE);
        	shadowedText.draw(canvas, paint, mLeftComment, Color.BLACK, leftX, topY);
        	shadowedText.draw(canvas, paint, mRightComment, Color.BLACK, rightX, topY);
`

apps4av, 2016-11-22:
The GPS will invalidate every one second or so. Unless if you are in sim
mode where you found refresh is every few milliseconds ?
Need more info here as it's a good find.

pasniak, 2016-11-22:
Right but if nothing changes, nothing should draw (for ex: we can check if nav display fields changed state). Not sure if that helps: I start the app, switch to Plan, and back to Map.

pasniak, 2016-11-22:
the overdraw viewer in Developer Options sounds like a good thing to try (http://blog.venmo.com/hf2t3h4x98p5e13z82pl8j66ngcmry/performance-tuning-on-android)

apps4av, 2016-11-22:
> /performance-tuning-on-android)

Good find.
Overdraw shows buttons are drawing and top status lines are drawing. Could
be optimized. But why buttons? Interesting.

pasniak, 2016-11-22:
Also, on Map the aircraft point is dark red (aka "you are doing it wrong" :).

Another cute option on my tablet is "Show screen updates/Flash areas of screen when they update".  When I set it, the map flashes every second (again this is when I am not moving, no GPS etc.) - not too bad; Find flashes every few ms, BUT the **3D**, **Plan**, **List** views are quite bad - they just flash continuously!

pasniak, 2016-11-22:
maybe the red ares are just overdraws over the map layer?  yet, when i go outside the area where I have the map (the black zone), the bottom buttons become only reddish, but top stuff (Simulation, Next, Dist Next) are still red.  The question is why are they drawn few times?

apps4av, 2016-11-22:
The black area is still images so it does not change drawing.

pasniak, 2016-11-22:
Here's how to see what invalidates a view:

http://stackoverflow.com/questions/10570366/in-which-condition-the-ondraw-method-of-view-to-be-called-repeatedly-in-android
> solved it by overriding invalidate() in my View subclass:
> 
> @Override
> void invalidate() {
>     super.invalidate();
> }
> Then I set a breakpoint inside invalidate() and waited for it to hit. When it did, I had a stack trace of the call, and the offending caller.

pasniak, 2016-11-24:
a) in Plan view constant redraw (blinking) of the webview is caused by
`mHandler.sendEmptyMessage(MSG_INACTIVE);`
in
`WebAppPlanInterface.timer()`
which then calls 
`mActivateButton.**setText**(getString(R.string.Inactive));`

setString invalidates the entire Plan page, and this happens every second.  Fixed in PR #254

b) I still observe CPU drain from Plan webView JS timers. This stops when I add 
`mWebView.pauseTimers()`
to PlanActivity.onCreate. Sadly, this also seem to stop new waypoints to be reflected on the plan view.  Need other ideas... maybe something along [Managing WebView state](https://books.google.com/books?id=hI8sBQAAQBAJ&pg=PA182&lpg=PA182&dq=webview+timers&source=bl&ots=IoQt1qWo7d&sig=k9J5uxG9i4QNI1OmWXxxYm3XkvM&hl=en&sa=X&ved=0ahUKEwjZ_dfsusHQAhUJOSYKHW-mC3M4ChDoAQg6MAY#v=onepage&q=webview%20timers&f=false )  in Advanced Android Application Development?

apps4av, 2016-11-24:
The app draws every 1 second. It used to be every time a GPS location event
was sent, but was improved after database query was found to be happening
too often. I think leaving it at 1 second is not bad. Its nor perfect but
it works without too much average CPU use.

pasniak, 2016-11-25:
It is not the map draw.  It happens in Simulate mode at **0kn** as soon as I click on Plan. This instantiates the Plan WebView which creates plan_poll_timer.  The parsing of the plan is up to 8% Kernel CPU every 1000ms (https://i.imgur.com/Oj4N31j.png)  

Kernel CPU usage is 1.8% if I comment out 
`var plan_poll_timer = setInterval(function () {loadplan()}, 1000);`
in plan.html (or set the pooling time to some very high number). 

The fix will be to disable plan_poll_timer when Plan is not in focus.

pasniak, 2016-11-25:
The plan pooling fix in PR #256

pasniak, 2016-11-25:
2 observations re: live fields (ETA, Time....) : 
1) we use a local calendars which are constantly re-allocated. 
2) String.format could be replaced with + or StringBuilder, [see](http://stackoverflow.com/questions/513600/should-i-use-javas-string-format-if-performance-is-important#)  

In my experiments Plan.updateLocation time can be shorten 50% from 150 to 70ms. 
Similar drawCornerTextsDynamic's time would improve.

pasniak, 2016-11-26:
What do you think about making this calendars static?  For example this calendar makes draw double slow, because all shapes check their expiry, and this buy gets re-allocated for every check.
`
    private static Calendar calendar = new GregorianCalendar();
    /**
     * 
     * @return
     */
    public static long getMillisGMT() {
        TimeZone mTimeZone = calendar.getTimeZone();
        int offset = mTimeZone.getOffset(System.currentTimeMillis());  
        return System.currentTimeMillis() - offset;
    }

`

apps4av, 2016-11-26:
Agree. Thats also another good find. Will get this from your PR.

apps4av, 2016-11-27:
Committed calendar change.

closed 2017-07-16 by pasniak