apps4av/avare #265: AVARE I/O

boa888, 2017-01-15:
Jan 14 2017
Been testing Avare for a 6 or 7 hours with no problem. Today I continually got a message "Avare IO has unexpectedly stopped" or something like that. I would restart but it you only run for a few minutes before I got the same message. 
 
I am running Avare from the SD card (class 10). Avare I.O is on the tablet (not SD card) and I have not cleared cache or data.
 
I do have but did not try iFlyGPS. Next time I will start it just to see the Stratux is still outputing data. I am willing to help. I am taking Instrumsnt lessons so I fly almost every day so I can provide testing feedback. 
 
Avare I/O 4.0.2
Samsung Galaxy Tab 2 Wi-Fi
Stratux with VK-172 Gmouse
 
After landing I started Avare IO and got messages about Battery Low: False

avcomware, 2017-01-15:
Hi.I posted your message on the Forum: https://groups.google.com/forum/#!forum/apps4av-forum

Post more info there about the Device Make, OS ver etc. in case others may be able to help.I tested Avare with a hand built Stratux for a few hours and I did not notice any problems, with a Tab 4, make sure you read the Forum about the Port you must use in the IO, the Avare does not default to it at start up. TV

boa888, 2017-01-16:
Samsung Galaxy Tab 2 7" Android 4.2.2
If I continue to have problems I will try my Tab 4

closed 2017-01-16 by boa888

apps4av, 2017-01-17:
For all IO app issues, please send a crash report generated, and if possible, record the ADSB data using the "record to file" check mark in the IO app. Then send the recorded file so your issue can be replicated here. Thanks!

apps4av, 2017-01-17:
To record a file, check "Save" button so that it becomes "Saving", then
send the file after crash from the path shown after the "Saving" button.

boa888, 2017-01-18:
I would love to help but I don't understand the instructions. I can "Save" but I have no idea how to locate the file.

apps4av, 2017-01-18:
May want to try the attach feature of email in phone to get to the file
listed in the "Saving" area.

boa888, 2017-01-18:
I found a no-nonsense FileBrowser so I will be able to email the file. At what point do I tap "Save"?

apps4av, 2017-01-18:
As soon as you start receiving ADSB packets. Thanks

boa888, 2017-01-18:
Worked great today and I the problem may be solved. My Stratux uses a VK-172 dongle. I noticed the LED was not blinking. Turned out it was not making good contact. I re-seated it and had no problem. I will fly again tomorrow and let you know if there is a problem. 

It does seem strange that the Avare I/O would fault on no GPS data especially since the tablet has internal GPS. And the error message was about a bad battery did not make since either. But it looks like I will join the "donor" class soon.

avcomware, 2017-01-19:
Hi.If you use an External GPS turn Off your device"s GPS otherwise you may have a conflict.I found out that my Internal GPS from my Samsung tablets works fine, but if you use external turn Off Internal. TV

boa888, 2017-01-19:
Always love to start with "I could be wrong". The situation was that I had GPS data turned on in the Stratux. Could it be that Avare I/O did not like the data stream because there was no GPS data. It could not be a conflict as the GPS dongle was disconnected hence no conflict. 
I built my Statux over a year ago and have never had a conflict problem with iFlyGPS and I think Avare works the same way. In fact Avare GPS setting "All Available" would lead you to believe it works the same as iFlyGPS. Furthermore in the GPS settings I checked the "Warn GPS Disabled" but I never got any such message.
I would bet good money that the problem was that Avare I/O could not deal with an invalid datastream. I am not a coder but perhaps someone might look at what triggered that particular error message. (Battery Low: false) and sent it into a do loop.

boa888, 2017-01-19:
No problem today so I am sure the problem was that the GPS dongle was disconnected.

apps4av, 2017-01-19:
Yes the battery low message is no big deal. It wont crash on something it cannot handle. But it may crash if bitstream from air has errors. Thats what I wanted to find with recorded file. Reopen if seen again with recorded file.