apps4av/avare #266: possible to lose tabs behind soft keyboard switching from Find

pballer2oo7, 2017-01-16:
1. switch to Find tab
2. tab search textfield to pull up soft keyboard
3. tab bar appears above soft keyboard, tap Map

the tab bar goes back to the bottom of the screen, hidden behind the soft keyboard which does 
[image]
[image]
[image]

not dismiss

reddn, 2017-01-16:
Isn't this normal android behavior.

pballer2oo7, 2017-01-17:
we should be dismissing the keyboard, if it's open, when we change to Map tab

apps4av, 2017-01-17:
Fixed release 7.6.6.

closed 2017-01-17 by apps4av