apps4av/avare #267: fullscreen mode

pballer2oo7, 2017-01-16:
I'd like to be able to turn off the Navigation Bar and the Status Bar to take advantage of full screen height, particularly while using approach plates.

The api 19 way of doing it is pretty sweet. Both of the bars autohide but can be pulled up with a swipe near the screen edge.

We can put a check if the system is running api 19 _and_ the user enables the feature. We can set this mode on the view associated with the MainActivity. _Nothing changes_ if the user is running 18 or older or does not enable the feature. Is this an acceptable approach?

If this is done, some other small UI tweaks should be made as well:

-  padding or lowering the TextField on the SearchActivity (because when using stable immersion, the search textfield is partially obscured by status bar when a soft keyboard is raised)
- present DialogInterfaces in an immersive way (would help with ux fluidity since we're using Dialogs to present lists of choices)

pballer2oo7, 2017-01-17:
need to fix WXB, Plan, and List.

When you open one of these tabs, and then enable the soft keyboard, the stable immersion prevents you from being able to scroll to the bottom of the screen and filling out the entire form.

JohnW4SBA, 2017-01-17:
> bars autohide but can be pulled up with a swipe near the screen edge.

Seems like a cool optional feature that's Off by default (I'm a big fan 
of starting newbies in "simple" mode).  Are there any handles or buttons 
in the corners when auto-hid, or could some be added?  Seems like a good 
safety feature to have something as a reminder at the left edge corner 
where the bars would normally be.

pballer2oo7, 2017-01-17:
I agree completely with you about default behavior, JohnW4SBA. I have it defaulted to off in my branch. 

I haven't added any bars or handles. immersive mode does teach the user how to use it (how to recall the bars) the first time it's enabled. we can also add it to the tips and help.

apps4av, 2017-01-17:
Have you tested thoroughly? We do not have a test team so developers make
sure their changes are thoroughly tested. Once a PR is merged, and
released, and serious issues are found, it will be backed out with longer
code merge cycle from that developer for future PRs. Thanks for your help.

pballer2oo7, 2017-01-17:
@apps4av, agreed. I won't open the PR until it's ready.