apps4av/avare #271: GPS coordinates added first when creating a plan via copy/paste

EasternPA, 2017-02-20:
**Summary**
Something changed recently. When I copy/paste a list of waypoints into the Create New Plan field and click the "Create" button, GPS coordinates get added to the top of the plan instead of where they are shown on the list that I pasted into the field.

**Details**
I have manually built flight plans that my flights normally take for my regular commute and I include the GPS coordinates for the decision point over the landing runway, followed by one or two more waypoints for a missed approach. Now when I copy/paste any of my curated plans, the GPS coordinate for the decision point at the arriving runway is the first waypoint, followed by the entire route from the departing airport to the arriving airport. This is reflected in the map, with the first waypoint being the decision point over the runway I expect to arrive followed by the entire route from the departing airport to the arriving one.

**Example**
See these waypoints for KPHL 27R, which includes the GPS position for the decision point as well as the waypoint for a missed approach

ENZEW DUCVA MRTIN JALTO 39.8752&-75.2229 MXE

Copy and paste that into "Create New Plan" and you will see the GPS coordinates appear at the top of the Plan. This happens every time I copy/paste my curated plans into the Create New Plan field.

**Workaround**
For now, I have to copy/paste my curated plans up to (but not including) the decision point. Then I go back and copy/paste the decision point into "Add Waypoint to Plan", click Search and then Add the GPS position to the tail end of the plan.

Thanks!

avcomware, 2017-02-20:
Hi Steve.I can confirm that what you are describing occurs in my devices also. I am not sure when that started since I don't normally use that procedure but it will take a few days to be looked at and corrected.
I have one suggestion, you may be aware of it, but you can create your FPlan, as you described, and just before you save it, use the Dn button at the bottom right below the plan and move the coordinates where you want and then Save it?
I know this will Not fix the problem but it may make it easier until it gets fixed? TV

EasternPA, 2017-02-20:
Thank you for reviewing and confirming my report so quickly. It's only been doing this for about 3-4 weeks after working reliably for several months. I also appreciate the suggestion to save the corrected version of each flight plan, but I think I'll stick with my current workaround for now. I normally have 50+ waypoint flights that are customized based on the winds at each end and the possibility of weather along the way.

I often have to customize the plan with less than an hour to go before the flight and I would have way too many versions to sift through to find the exact one that was filed. It's easier to just staple together the departure procedure to the middle section(?) to the arrival procedure and then the approach procedure and final approach.

Another option for me to consider is to replace the GPS coordinates for the decision point with the destination airport ICAO code. That will at least show me the runway extensions, making it a bit more interesting on the display. I'm just a passenger sitting in coach, so I'm not your target demographic. I just wanted to point out something I've noticed very recently and I appreciate your speedy reply.

avcomware, 2017-02-20:
Hi Steve.
In Find, you can also, create xx.xxxx&-yyy.yyyy  > Label> add to Plan  your coordinate, but it will place it in a location that you may still have to move it. TV

EasternPA, 2017-02-21:
Thanks for the additional follow-up. Another suitable alternative would be to allow us to append saved plans to the end of the current plan. That way, I could save the departure and approach procedures that frequently come up and just add them to the current plan as needed.  When flying back to KPHL, for instance, my flights typically use the BOJID2 or PAAT2 arrivals. I have curated the most commonly used waypoints for BOJID2 and PAATS2 in Google Keep and I've also curated the most commonly used final approach to 27R.

Same for departing KDFW. I have curated the two primary departures depending on a North or South departure. As my departure time approaches, I take the filed flight plan, expand the waypoints for the departure procedure, add the intermediate waypoints, then add the expanded waypoints (FlightAware calls them 'decoded') for the approach into KPHL and then add the final approach into 27R.

If I could save all of those individually and append them one after the other to the current plan, that would replace the manual work I'm doing to assemble each complete plan using Google Keep. Otherwise, if I can only replace the current plan with a saved plan, then I will need to copy/paste my curated segments into the Create Plan field.

apps4av, 2017-02-28:
Thanks for finding. Fixed for release 7.6.8. If found in 7.6.8 or above, reopen.

closed 2017-02-28 by apps4av

EasternPA, 2017-03-01:
Thanks for the quick turnaround on this, and for opening #272 based on my suggestion to give us the option to append saved plans to the end of the current plan instead of only replacing it. I look forward to seeing that feature get added.