apps4av/avare #276: [Request] Show traffic on approach plates

EasternPA, 2017-03-01:
Issue #104 has several requests contained within, many of which have been achieved since the issue was opened a couple years ago. I was going to ask about showing ADS-B traffic on georeferenced approach plates, but that request was already shown on the list.

Any chance you can review that issue and comment on what has been achieved and which requests you plan or don't plan to address? The issue basically got zero traction when it was opened.

gustafson, 2017-03-15:
Hi Steve, this is an interesting concept which Zubair is much more well
placed to answer.  However, you can now display the plate as on overlay
over any of the other maps (sec, tac, ifr, etc) and when doing so I
assume traffic would display just fine.  This process is similar to
overlay of weather radar on the maps.

Pete

EasternPA, 2017-03-15:
Excellent, how do I do that? I don't see it in Preferences -> Display, Preferences -> UI Configuration, or on the Plates tab.

Thanks!

apps4av, 2017-03-15:
I think plate is best left uncluttered because it already has a lot of info
on it. I guess you want it to shoot practice approach on vfr as traffic is
cleared by ATC in imc. Try it with Map layer as Gustafson remarked.

EasternPA, 2017-03-15:
Thank you, I didn't realize this option was on the Layer menu. I checked it out and got it working. FltPlan Go (and perhaps others) takes the same approach. I try not to be too annoying around here because I'm simply an enthusiast, not a pilot. I am a road warrior and fly on two to four airplanes a week. As more airports deploy ASDE-X, more pilots are leaving ADS-B activated while taxiing. I usually have the Airport Diagram up while taxiing around and thought it would be fun for me and useful for pilots to see other traffic on that plate.

As an aside (and again, I'm not a pilot and don't think like one), overlaying the plate seems like it over-complicates the Map page. Are pilots normally that focused on executing the approach that they (you) don't necessarily want to know if you are too close to another aircraft in the pattern? In trying to think like a pilot, I keep thinking that when I pull up the final approach plate, I would want to see other aircraft on that plate without all the other content found on the map view. That doesn't look to be the case.

Thanks again for reviewing this with me!

gustafson, 2017-03-15:
FYI This is probably best discussed on the forum rather than github. 
This is a pretty narrow audience.

But in answer to your question:

When you are cleared for an approach in an IFR environment, then you
have guaranteed separation (by ATC) from other traffic.  So you aren't
worried about (and don't want to be distracted by) other traffic.  The
approach plate is plenty busy with information.  You've got enough going
on in the cockpit that another piece of info might overwhelm rather than
help.  It is my view that (in real IFR conditions) traffic should not be
displayed on the plate.

The exception is the "cleared visual approach" in visual conditions, but
this (technically still IFR clearance) doesn't have an approach plate. 
You might have an approach up for reference... and in this case traffic
might be usefully displayed.  But the far more stressful approach is the
approach when in actual ifr conditions and I'd prefer to design the
system to that.

Just one perspective... hope it helps.

Pete

On 03/15/2017 02:54 PM, Steve J wrote:
> As an aside (and again, I'm not a pilot and don't think like one),
> overlaying the plate seems like it over-complicates the Map page. Are
> pilots normally that focused on executing the approach that [...]

apps4av, 2019-03-15:
This is accomplished by overlaying of plate on Map. Closing.

closed 2019-03-15 by apps4av