apps4av/avare #306: Add Save GPS Location on Tools Page

apps4av, 2017-06-27:
Add a "Save" current GPS position button on the "Tools" tab page.
Log that position to a user waypoint list (maybe just a text file, which can be backed up and won't be affected by updates) with a default ID, like WP1, WP2, ..., with the ability to later edit that ID, which is why a text file.  This user waypoint file should be accessible by ID in Flight Plan and Find utilities.

Using a text file would allow us to manually add user waypoints offline, ie. off airport landing spots, private air strips, points of interest given to us by others, etc.

I fly an ELSA off of my property which would make this real handy.
Also it would allow inflight storing of locations of interest for returning to later by air or land, especially in search operations.

JohnW4SBA, 2017-06-27:
> Add a "Save" current GPS position button on the "Tools" tab page.

Hurrah!  This will be a popular feature, and I celebrate the addition.

Thanks :)

apps4av, 2019-03-15:
This has been fixed by addition of waypoints in the Find page.

closed 2019-03-15 by apps4av