apps4av/avare #308: [Question] why there's no line to first waypoint in a Plan?

pasniak, 2017-07-16:
After plan is activated. Is it bug or feature?

apps4av, 2017-07-17:
*Its not a bug. Where you start from is not a way point that distinguishes
it. You can start from anywhere, any airport.*

pasniak, 2017-07-17:
CDI does not work then on this "direct to the first plan waypoint" leg? I simulated few plans and CDI indications do not make much sense to me...

avcomware, 2017-07-17:
Be more specific on the Plan, Location, procedure how you tested that. I can test in a sim.

pasniak, 2017-07-18:
LEt's start with a simpler case.

I started the GPS sim at 42.14 -71.22 hdg 45 spd 290 and Fly To Destination. 
[image]

Then I selected destination BOS (no plan):
[CDI-full-left-2017-07-18-064052.zip](https://github.com/apps4av/avare/files/1155616/CDI-full-left-2017-07-18-064052.zip)

CDI white triangle is pegged to left.  (BTW: It is also unclear from GUI or Help what "one tick" is and whether it is in degrees or miles to distinguish from CDI in top map display which must be in miles)

"The CDI is operational when you have a destination set. The needle indicator at the bottom will move left/right of center based upon how off course your location you are. If you deviate more than one tick left or right then the indicator changes color - RED if you are left of course, GREEN if you are right. To correct, steer the aircraft in the direction of the arrow to bring it back to center"

I let the plane get to dest and fly between the two points and debugged CDI.java:
mDeviation is (for example) 0.020753605185033695
mBarDegrees is 0.5
mDspOffset is -280

avcomware, 2017-07-18:
Hi.
My procedure, typical for real life:
I placed the acft at KSNA aiport.
Long press on a GPS point and select D.
The CDI was centered in Green color.
As I was flying to the South the pointer changed to Color Red and the pointer shows needed right correction.
As I continued on it performed as expected.

That said I think that the colors are not very intuitive / logical, when I compile my own version l select Green for Center, Red for Right and GRAY for left. For the VDI I use Blue High, Green center Red if you are low.
I also make changes to the List to accept 500 lines, that are needed for detailed Checklists for students mainly and a few
others.TV

[image]

avcomware, 2017-07-18:
Hi.My procedure, typical for real life:I placed the acft at KSNA aiport.Long press on a GPS point and select D.The CDI was centered in Green color.As I was flying to the South the pointer changed to Color Red and the pointer shows needed right correction.As I continued on it performed as expected.
That said I think that the colors are not very intuitive / logical, when I compile my own version l select Green for Center, Red for Right and GRAY for left. For the VDI I use Blue High, Green center Red if you are low.I also make changes to the List to accept 500 lines, that are needed for detailed Checklists for students mainly and a few 
others.TV