apps4av/avare #313: Approach plates and AF/D airport diagrams out of date

hmehendale, 2017-08-17:
I noticed today (after today's charts update) that the approach plates and airport diagram are being picked up for the 22 June - 20 July cycle. This happens both on my phone and tablet, so it's not isolated to one device. Is it bad data coming from upstream, or is Avare updating things incorrectly?

The IFR Low plates themselves seem to be updated (e.g. in Norcal area, MXW and ECA VORs are showing shutdown, which is correct).

apps4av, 2017-08-17:
It tests ok. Reinstall avare in case you selected a wrong cycle in
preferences.

apps4av, 2017-09-14:
Non issue.

closed 2017-09-14 by apps4av

vinodkutty, 2017-09-18:
I was able to reproduce this issue and have some additional observations + resolution. I noticed it a week ago on Avare 7.8.0, and today on 7.8.1. Chart cycle in preferences is "Current" and I've never changed this.

I really hope re-installing Avare is NOT considered a normal approach to resolution of such issues, so I spent a little time fiddling with Downloads. I checked and all A/FD Images were all `1710 ( 09/14/2017-10/12/2017` ). 

But any Airport Diagram I pulled up via long-press on an airport and selecting "Plate" still said `EC-3, 22 JUN 2017 to 20 JUL 2017`. I deleted and re-downloaded, but that didn't make a difference. For example -- `KBMI`.

I then deleted all my A/FD Images without downloading, but I could still pull up the plates. So I don't know where Avare was looking.

I then realized I should really be looking at "Downloads->Plates/d-TPP/Ads". Went there and selected "Illinois" -- which was not selected (for that matter, nothing was selected) -- and downloaded it. That now pulls up the `EC-3, 14 SEP 2017 to 12 OCT 2017` airport diagram for `KBMI` (and others).

So, I don't know for sure when this happened on my device since I haven't been using Avare regularly for a while. The fact that I hadn't selected any Downloads under "Plates/d-TPP/Ads" suggests there is a now obsolete download selection or destination directory that has older downloads that haven't been deleted (for 22 JUN - 20 JUL), and is still referenced first if I don't have the newer / correct ones downloaded.

Compounding this was a mis-understanding on my part about what to select in the Download screen. I suspect others have run into this (or aren't paying attention to the dates in the Airport Diagrams ).