apps4av/avare #315: Add config for default A/FD content

pballer2oo7, 2017-10-04:
As a user, I'd like to be able to set what content is shown on the A/FD tab by default, so that I can view my preferred content (whether it be Avare DB content or A/FD images) with one touch.

I am accustomed to the FAA A/FD layout so my workflow to view A/FD pages is:

1. long press airport
2. select AF/D from popup
3. (on A/FD tab appears) press A/FD select box at top of screen
4. (on A/FD select list that appears) choose sc_384-0 (just an example) 

I'd prefer to be able to set an option so that the FAA A/FD image appears by default after the airport long press, eliminating steps 3 and 4. The config option might fit under **Display | Default A/FD Content** with two mutually exclusive options **Avare DB** and **FAA A/FD Images**.

pballer2oo7, 2017-10-10:
Added a ListPreference under **Display | Default A/FD** with two options: **Avare Data** and **FAA A/FD Images**. The default is Avare Data (so all user workflows are unchanged unless they want to take advantage of the feature).

If **FAA A/FD Images** is set, then when the user goes to the A/FD tab, the setViewFromPos method is called with index=1 (since the first AF/D image file is the second element in the array list). This method already has a fall back mechanism which causes us to simply display Avare's parsed A/FD data if no A/FD images are downloaded for that airport.

Testing notes:

1. set FAA AF/D Images as default
2. long press an airport on map tab for which A/FD images are downloaded and press A/FD; verify the first downloaded A/FD for that airport is shown (not the parsed data)
3. go to the **Find** tab
4. search for and select an airport that has downloaded A/FD images (this sets that airport as destination)
5. switch to AF/D tab and verify A/FD image from step 2 is still displayed
6. select the airport from step 4 on the A/FD tab, verify an FAA A/FD image is shown (not the parsed data; this test is important because we use specific logic to load A/FD for the airport that is the current destination)
7. select the airport from step 2 on the A/FD tab; verify FAA A/FD image is shown
8. select the same airport from step 2 again on the A/FD tab; verify FAA A/FD image is shown (this test is important because we use specific logic to load A/FD for the airport that is already being displayed in A/FD)
9. long press the same airport from step 2 on the map tab and press A/FD; verify that FAA AF/D is loaded
10. verify you can switch between AF/D images and Avare A/FD
11. step 2 should be done with an airport from each AF/D volume
12. set Avare Data as default and repeat steps 2-9, this time verifying the normal Avare parsed data A/FD is shown by default

apps4av, 2017-10-10:
Sounds good. Will add in next release.

pballer2oo7, 2017-10-11:
here's a small "bug" due to how we check last afd destination; not sure if i want to fix as it will be a very rare occurence:

1. with afd default set to Avare Data, long press an airport and tap afd; Avare Data afd will appear
2. change default to FAA images
3. long press same airport and tap afd; Avare Data afd will still appear for that airport until another airport is selected

run25I, 2017-10-20:
1. all test [steps](https://github.com/apps4av/avare/issues/315#issuecomment-335346331) PASSED
2. step 13 questionable - the Default A/FD check box is somewhat unclear - what exactly is the default?
3. if we use a check box the label should indicate what would be used as the default eg. "Use A/FD images as default"
4. alternatively we could use the menu design in 356897e