apps4av/avare #316: Transform location button into a unified GPS status area

pballer2oo7, 2017-10-13:
Currently we present information about GPS in a couple places: the location button (turns green/black for track/north up) and the status area (red text for "Simulation" and "No GPS Signal").

Instead, we should use the location button (the center button on map, A/FD, plate, and other screens) to present information about GPS in one location in a unified fashion.

First step: add "SIM MODE" and "NO GPS" icons and remove the red text from the status area (this will also pave the way for some other improvements: #172, #173, #231).

NO GPS mockup:
[image]

SIM MODE mockup:
[image]

pballer2oo7, 2017-10-13:
For the NO GPS icon: I'd like to add perhaps a red "X" or some other graphic to make it very noticeable that GPS signal has been lost. 

For sim mode: the user is already going to know they're in sim mode so just a simple white text reminder is sufficient.

n251ea, 2020-06-08:
what I think would be helpful especially to those that use track up for charts is place a north pointer inside this Icon.

another possibility would be the ability to move it to a different part of the screen (eg drag an drop) as small screens might have an issue with the location.

pballer2oo7, 2020-06-18:
I like the north up pointer idea. Drag and drop relocation of the button is a good idea too but will be challenging given the current structure of the screen.