apps4av/avare #318: Supported Device?

DemonRem, 2017-10-28:
what device support this application?

JohnW4SBA, 2017-10-28:
> what device support this application?

Avare and all Apps4Av apps follow official Android guidelines and 
practices.  So Avare runs on most Android devices that were built within 
the last few years.  I don't think there's a definitive list anywhere. 
Lots of variables affect what devices will work with Avare including 
device specs, Android version, updates, and even other apps installed 
(many don't follow Android guidelines, so they can conflict with other 
apps).

The simplest way to see if and how Avare works on a device is to install 
and test it.  That's what I did in Costco before buying an SM-350 tablet.

DemonRem, 2017-10-28:
I see thank you jhon

gustafson, 2018-08-14:
Closed not a bug.  Questions more appropriate for the forum.

closed 2018-08-14 by gustafson