apps4av/avare #319: ADS-B traffic shows Pressure altitude with relative altitude tag

GitUser275, 2017-11-17:
Test configuration:
Startux v1.4r3 with dual SDR but no GPS
Moto G android phone with integrated GPS but no baro sensor
Avare ver 7.8.5

Test condition:
On ground with Moto G providing GPS position and Stratux providing an ADS-B data stream that does not include the periodic ownship report. (Statux should provide ownship report with no GPS connected and this issue has been logged by Statux)

Observation:
Traffic is shown with a tag in relative altitude format but the displayed altitude value is target pressure altitude not target relative altitude.  The following screen shots were taken in the order displayed.  They show that AAL1246 was flying at FL360.  The target tag for AAL1246 shows +36000'.  Ownship GPS altitude is 1250 ft.   For this condition the target tag should either have been 36000 PressA or +34750.

[image]
[image]
[image]
[image]

apps4av, 2017-11-18:
Thanks. It's obviously wrong. Is it using internal GPS? What are the
preferences?

GitUser275, 2017-11-18:
Not sure if I should reply to this email or reply on Github.

As started in the original post:

Moto G android phone with integrated GPS but no baro sensor

On ground with Moto G providing GPS position

Not sure what preferences you need.  GPS preferences are:   Short period update - enabled; Warn GPS Disabled - enabled; GPS position source - All available (but there is only one available).

So - Internal GPS was in use and was not flagged invalid.

However, see second post to issue 319.

Andy