apps4av/avare #320: Databases updated, but showing outdated charts

jonhnet, 2018-01-20:
I have three devices: one phone and two tablets. I did a database update on all three last night.
Before my flight, my instructor and I looked at the KRNT GPS-Y on the phone. It's version 17AUG17, and it agreed with the version in his Jepp binder.

During the flight, my tablet showed a different chart (11DEC14) -- I noticed because the new hold at LUTSY  is missing. I thought "uh oh, I must have made a mistake", so I swapped out the tablets. But the other tablet is _also_ showing the 11DEC14 version.

That's scary! And surprising, since I updated all the charts. They're all showing green.

Oh -- weird. Under Plates/d-TPP/ADs/Washington, there's a (?) symbol. So the tablet has an old version of the chart, but is silently letting it get stale! YUCK! The phone has the correct green [check] symbol.

This is a significant UI/safety issue: I thought I had charts, and Avare was showing me charts -- but it was showing me charts from a stale download. Yikes.

apps4av, 2018-01-20:
Whats the app vesion you are running.

As of today charts should be 04JAN18. 

11DEC14 is 3 years old, and 17AUG17 is more than 4 months old. You need to keep updating your charts.
None of the situations you describe above will give you the latest charts.

closed 2018-01-20 by apps4av

jonhnet, 2018-01-21:
I think this is actually a real bug plus a feature request. The tablets are the feature request: the chart-selection UI made it very easy to think I had charts selected, but "update" was a no-op.

But there's a real bug, too. In the linked screenshots, notice the following:
https://photos.app.goo.gl/F9VzY24XcYBUlAe73
Avare is version 7.8.6. Google play thinks it is fully updated; it says "open", not "update."
The help page indicates that expired charts should appear red in the list.
The Washington TPP plate entry is green and checked in the chart download page.
Yet the plate page is still showing 17AUG17.

Once when I clicked "Update", it showed "Washington TPP", and then completed, and still is in this (stale) state. Most other times, it updates the weather/TFR package, and everything goes back to green. But it's still showing a 17AUG17 chart. There's really a dodgy bug here where the chart page claims I'm up-to-date; "update" does nothing, and the chart is enthusiastically out of date.

apps4av, 2018-01-21:
Hello Jon

The version is correct in the picture you showed. It says the plate expires
in Feb 2018. You are looking at last amended date.
The bug to remain closed because the picture shows things are OK.

https://photos.app.goo.gl/F9VzY24XcYBUlAe73

<https://github.com/apps4av/avare/issues/320#issuecomment-359222453>, or mute
the thread