apps4av/avare #321: Logo Designer

dibon-github, 2018-03-29:
Good day, I would be willing to collaborate and make a new logo design for your app. Pls do reply if you accept my offer. I do work for free, dont worry sir. I would do anything for the benefit of this project.

apps4av, 2018-04-02:
Hello
Thanks for the offer. What do you have in mind and why is the current logo
not acceptable?

dibon-github, 2018-04-03:
I do have something in mind, a modern approach for your logo because it looks a bit old. Would you mind giving me a chance to give you a design tomorrow? and see If it fits your project?

reddn, 2018-04-03:
In this guys defense, he may be only looking at the play store icon, and
not the launcher icon, which are different.

See attached baa

reddn, 2018-04-03:
I do think Avare should change the play store icon to the launcher icon.

reddn, 2018-04-03:
Sorry for the multiple emails.  Need to clarify what I said, it sounded
right in my head when I first typed it.

I think Avare should make the play store icon the same as the current
launcher icon.

dibon-github, 2018-04-03:
yah maybe sir, thanks for clarifying that to me

dibon-github, 2018-04-03:
if ever you need help for anything feel free to contact me

apps4av, 2018-04-03:
Hello baa
Please describe what you want in the new icon. Our icon is our identity but
if you have a great idea we want to hear about it.
Thanks for offering.

JohnW4SBA, 2018-04-03:
> a modern approach for your logo

I'd like to see that and I very much appreciate your offer, but I'm 
unclear.  Is there some reason you might prefer not to design a new logo 
unless you are assured in advance that it will be used?

Might you be willing to provide some rough samples that could be created 
while minimizing your effort, so that others here could provide feedback 
or requests that you could consider for implementation?

dibon-github, 2018-04-04:
[image]
this is whats appearing upon clicking the link. Pls just send the image of your current logo. And Ill do make a design and send it tomorrow. It doenst matter if it gets used or not. Ill just do my best and see if I suite your taste.

dibon-github, 2018-04-04:
At first the logo I saw on your playstore was I thinking, not knowing that your current launcher icon is different from the logo displayed here. sorry for that misunderstanding.
[image]

gustafson, 2018-08-14:
Marking as closed as the originator appears satisfied.

closed 2018-08-14 by gustafson