apps4av/avare #322: Distribute through F-Droid

picsi, 2018-05-20:
It would be excellent to have this also distributed through https://f-droid.org/

picsi, 2018-08-16:
Maybe @rudloff would know?

picsi, 2018-08-16:
@gustafson do you think you folks could build a gradle flavor without play-services?

JohnW4SBA, 2018-08-16:
pizzaiolo wrote:
 > @gustafson do you think you folks could build a gradle flavor without 
play-services?

I understand that you and perhaps some other users would very much 
appreciate that, yet want to state my opposition for two reasons:
1. Anything that adds even a minute of work for the Core Dev Team is a 
minute not spent on charts, maintenance and improvement.
2. I appreciate the modest margin of safety provided for the vast 
majority of our users by gPlay.

There are probably many other reasons that the CDT could outline, but 
I'm hoping they won't need to take the time for that.  Especially right 
now as the deal with the aftermath of the chart issue and hopefully get 
back to work on version 8 release.

So might you be willing to withdraw this request until things settle 
down and we're all done with Summer vacations?  I'm confident your 
request is on their minds, and they'll respond when convenient.

Thanks for your clear request, and for considering my response.

Happy Flying -j

picsi, 2018-08-17:
Thanks for the patient reply! I understand this might not be a priority, but I wanted to check anyway. I hope they consider this idea in the future. For now I'm closing this.

closed 2018-08-17 by picsi

JohnW4SBA, 2018-08-17:
> I hope they consider this idea in the future.

I'm glad you brought it up on behalf of others who might like it. 
Thanks for your participation in the Avare project. :)

Happy Flying  -j