apps4av/avare #324: Stratux AHRS Support

pophop99, 2018-07-18:
Will you be providing support for your PFD feature leveraging stratux?

cyoung, 2018-07-18:
A webview might work.

[image]

JohnW4SBA, 2018-08-14:
> Just thought your avare PFD page could leverage the same data

Makes sense to me, but I've no clue about the tech details involved.  I 
don't know of anyone working on it, but having the html available would 
seem to me quite helpful in mushing the StratuX/AvarePFD code together 
somehow.  Hopefully it will happen.

Happy Flying  -j

cyoung, 2018-08-14:
The HTML for the AHRS display in Stratux? It's available [here](https://github.com/cyoung/stratux/blob/master/web/plates/gps.html).

It is rendered using JavaScript-generated SVG, [code here](https://github.com/cyoung/stratux/blob/master/web/plates/js/ahrs.js).

Let me know if you need anything else.

digivation, 2018-08-14:
Is the information from https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md `getSituation` useful? Seems like it would be better to drive the PFD displays already in the app, instead of adding another webview, unless I'm misunderstanding something.