apps4av/avare #346: Migrating from WebView?

sergeant-wizard, 2019-09-21:
Hi, I'm a big fan of the project.

I was wondering if we could switch from WebView based interfaces to native Android UI elements.
Or is there a reason to stick with WebView like supporting old devices?
I think more users will like the new looks.
It's going to be quite a major refactoring but I'm happy to help!

gustafson, 2019-09-27:
Thanks for your interest. As far as I am aware there are no reasons not to do this. However, as you mentioned this would require a fair amount of work. (There could be things that would be higher value for the time commitment.). Nevertheless it is likely a great way for somebody to learn Android programming.

I don't think that this is going to be a top priority for the core developers in the near future. Hence is a an opportunity for you to make a great impact? Give it a shot and let us know when you are ready for test.

N520TX, 2019-10-01:
I started tinkering with this a little bit ago with the W & B tab. This is what I have so far. Working on connecting the save/load logic to support legacy data.

[image]

apps4av, 2019-10-01:
Nice.
No graph?
Thanks

N520TX, 2019-10-01:
Ya ya ;o) ... graph and help pending.