apps4av/avare #349: Vertical desent optional airport elevation

Wme92, 2019-09-29:
Allow vertical desent option to select airport elevation instead of default pattern altitude

jwr618, 2019-11-10:
Airport elevation should be the default. If you are planning your descent to be overhead the airport at airport pattern altitude you planned incorrectly. You should be at pattern altitude BEFORE you arrive in the pattern. This means one has to fly lower than what Avare is displaying if pattern altitude is defaulted. Not a good thing to teach; fly below glidepath. But how much below? Depends on distance. So this leaves the pilot guessing and possibly diving into the airport at the last minute.

If airport elevation is defaulted, ~3 nm for prop aircraft and 5 nm for jet before the airport you would be at pattern altitude for a 3 degree descent. For a 2 degree descent, 5 nm for prop and 7 nm for jet.

Airport elevation is the safer and better option than pattern altitude.

saabnut, 2019-11-10:
This is a non certified supplemental information product. Assuming the published pattern altitude is being used, I see no reason to change to airport elevation. At some point one must transition to flying the airplane by outside reference. It's safer to be high in piston singles, and in jets, I don't think Avare is the proper tool for setting up  the approach.

jwr618, 2019-11-10:
Saabnut, I do not understand your argument.

"This is a non certified supplemental information product." - True. Relevance?
"Assuming the published pattern altitude is being used, I see no reason to change to airport elevation." - Opinion based on?
"At some point one must transition to flying the airplane by outside reference." - True if set for airport elevation or pattern altitude. Relevance?
"It's safer to be high in piston singles, and in jets" - False for both. Safe is being "on" and not high or low. I guess you never flew heavy aircraft where you have to lose altitude and slow down. Being high is just as unsafe due to the maneuvering required to get back "on".
"I don't think Avare is the proper tool for setting up the approach." - Who said anything about setting up an approach?

How about letting users decide.

JohnW4SBA, 2019-11-10:
> How about letting users decide.

Guess you're upset about the decision to go with airport elevation?  As 
a VFR SEL pilot, in a real emergency it would be nice to arrive at 
pattern altitude with maneuvering and traffic avoidance altitude.

I agree it would be nice for anyone with a different preference to have 
an option.  I hope you can understand that there are many factors 
involved.  A few more that come to mind in addition to those already 
mentioned are:
*Avare is a free app built by volunteers.
*Volunteers do what they want, when they want.
*Friendly collaboration with volunteers is helpful.
*Avare is intended to be small, fast, and widely compatible.
*All changes to the code are time-intensive.
*Avare is open-source, so those seeking any change not being actively 
addressed by the Dev Team volunteers are welcome to make their own 
changes to the code to recompile for their own use and submit it for 
inclusion in a future release.

Thanks for sharing your request and outlining your concerns.

Happy Flying  -j

jwr618, 2019-11-10:
Hi John,

Understand all. I'm not upset; I just don't use the vertical descent profile because it only displays prop pattern altitude. I have flown jets with 4000' AGL overhead precautionary engine out emergency descent profile so it would be nice to have a user selected option.

I like Avare and I am grateful for the program. Thanks to the Dev Team for all they do! :)

saabnut, 2019-11-10:
Sorry, I did not intend to argue, only comment, and apparently punctuation failed:  
Should have read:
"It's safer to be high in piston singles."
"In jets, I don't think Avare is the proper tool for setting up the approach." (Visual or instrument, both are approaches.) I agree with jwr618 that ON is safest for jets.

91.129e3 applies to operations at class D and therefore C and B airports.
(e) Minimum altitudes when operating to an airport in Class D airspace.
...
(3) Each pilot operating an airplane approaching to land on a runway served by a visual approach slope indicator must maintain an altitude at or above the glide path until a lower altitude is necessary for a safe landing.
For this reason my opinion is high is safe. 
Again this is my opinion only, no warranty as to relevance or the risk management of others.

Wme92, 2019-11-11:
As most know and practice, aviation safety must always be one of the primary concerns.
Arriving at a non towered airport at times is an amalgamation of hiwing and lowing aircraft, with the primary concern for safety, the hiwing should never get down to Pattern Altitude until they have arrived at the airport; while the lowings should be at Pattern Altitude 3 to 4 miles before arriving at the airport; thus providing the greatest opportunity to see and avoid each other; unless of course, the primary concern is falling off the couch. Adding to the safety, jets know to stay high and wide and not to cut in front of other traffic, so we can all hopefully safely co-exist. Because of possible
 radio or other problems even at towered airports in the Pattern hiwings should stay a little hi and lowings at PA for safety.

The Initial concern of this issue was when cleared straight-in to a towered airport; it would be convenient if Avare had the option of switching from the default (for safety purposes) PA to AE. The controllers are usually trying to work the straight-ins into the flow of traffic and are NOT planning on a straight-in showing up at the approach end of the rwy at PA wondering what type of rental cars are available.

My understanding is this issue was previously discussed, with the conclusion to provide PA for safety concerns; which is fine, having the default setting for every time Avare is started to be PA, thus allowing the choice of PA or AE. I don't see any reason any serious pilot would not want the
option to choose, especially when the default is PA for safety concerns.

Plllllease understand I really like Avare and from a non coder prospective a few little tweeeeks would provide nice improvements.

closed 2019-11-11 by Wme92