apps4av/avare #356: Drawing on the plate moves to other plates

n251ea, 2019-11-03:
When drawing on a plate, the drawing will display on any plate displayed, for example if I circle the departure procedures that are relevant to me and then switch to the airport diagram, the drawing will be on the airport diagram (or any other plate) too.

[image]
[image]

reddn, 2019-11-03:
Does the feature state the drawing will stay only on the selected plate?

n251ea, 2019-11-03:
I don't think it states anything, but carrying the drawing to other plates is pointless

reddn, 2019-11-03:
I think that depends on the use case.  If you are looking at a STAR and ATC
gives you a route and you use that feature to write it on there, if you
change plates or anything, you lose it.