apps4av/avare #360: Exchange Flight Plan Information with Avidyne IFDx40

vancinad, 2020-02-17:
Avidyne provides an SDK that allows plan information to be pushed and pulled from IFD-series navigators. Provide an option that allows Avare to exchange flight plans with the IFD.

vancinad, 2020-02-17:
I have an IFD440 and am willing to work on this, but would like some design guidance from those who understand the Avare code. Directly manipulating the Planv10 string seems like a bad approach, but I'm not sure what classes/methods in Avare would best be used.

n251ea, 2022-01-15:
Avidyne has been contacted regarding their SDK

apps4av, 2022-03-02:
A chat session will be better. I think the major portion is working on the interface. Once that's implemen then importing the flight plan is no more than calling the Plan class and adding waypoints to it.

vancinad, 2022-03-02:
By "interface", I think you're referring to the integration with the AviSDK, correct? If so, I agree. The challenge is that it's all C++, and I've been unable to get it to run on Android. This may be due to nothing but my inexperience. 

Designwise, I think the best approach will be to create a stand-alone "agent" app that handles communication with the IFD and uses Intents to communicate with other apps. This will keep the C++ stuff out of the Avare code base, and make the IFD Agent app useful to other devs as well.