apps4av/avare #366: Metar not displayed properly, droping the last digit from the altimeter setting

n251ea, 2020-06-08:
[image]

MMU always drops the last digit from the altimeter setting, my transponder and GPS display the altimeter setting properly (IFD440, NGT9000)

datafile from avare IO that this screenshot was generated from is also attached

[avare-out.zip](https://github.com/apps4av/avare/files/4746539/avare-out.zip)

gustafson, 2020-07-22:
Did  a little research... haven't fixed it but it appears to be related to this:

metar.rawText

In
ContentProviderHelper.java

n251ea, 2020-07-22:
I didn't notice when I posted the image but the last digit is also missing for the pireps, "C25B/TB MO" and "OVC01"

vancinad, 2024-01-15:
I'm unable to reproduce this issue. Can you, @n251ea ?