apps4av/avare #375: Ambiguity with ADS-B traffic targets

gustafson, 2020-07-06:
When viewing traffic, there is ambiguity of flight direction for the targets that are shown.  The target is indicated by a dot and a line.  Is the line a past or future projected flight path?  I suggest the line should be an arrow which would indicate direction with just a glance.

vancinad, 2020-07-06:
That is highly standardized iconography. IMO, no "creativity" should be applied.

JohnW4SBA, 2020-07-06:
> should be an arrow which would indicate direction

Interesting idea, but I don't recall anyone expressing confusion about 
the current dot-line. Easy to add an arrowhead, but an arrowhead would 
obscure a tiny bit more of the chart. Yet I like the length of line 
indicating speed, even though that obscures more of the map.

gustafson, 2020-07-06:
That is compelling if true, but it true I wasn't aware. Define standardized... is it documented? By whom? Where? Do pilots encounter this anywhere and receive training?

Here is a clear example of a more intuitive display:
https://flightaware.com/live/

FWIW, even if a standard did exist, there are plenty of standardized representations which are based on needs of older technology.  And standards still benefit from a key, which we don't currently have available in-app.

An example where modernization is a good thing.  Compare the following:
https://www.aviationweather.gov/windtemp/plot
https://www.windy.com/pois?43.779,-82.617,6,i:pressure

This feature request arose Saturday. I was cruising and passing through some visually pretty cumulus clouds (to the great excitement of my young son). The app had a target about 6 miles off my nose and +400 ft above me in the app. ATC didn't mention the target.  Most likely we were both headed the same direction at IFR and VFR altitudes. But it could have been somebody trying to drop down through a break in the clouds going the other direction. The display didn't update quickly. There was no apparent indication of speed and a 180 degree directional ambiguity (for me, though I've seen it many dozens of times when under less workload). The flightaware representation would likely have made me more comfortable during that 3 to 4 minutes. As it was, my workload increased assessing the line and its risk.  Its worth talking about an alternative.

gustafson, 2020-07-06:
OK if we're compliant with a line, but compliant and informative are not always the same.  FWIW the line doesn't help indicate velocity, rather it indicates relative velocity compared to other aircraft that are also visible.  If it is a solo dot and line, speed is essentially lost.  But I'm not sure how to fix that aspect.

See this (midway down the page) where FF provides a directional indication when you zoom in.
- https://ipadpilotnews.com/2015/10/foreflight-traffic-tips-get-ads-b-traffic/
It would be an improvement in my view.

And I would go further to make it look like flightaware + the line because you would have an indication of size (discerned by ADS-B reported type), plus direction, plus speed.

gustafson, 2020-07-06:
Can you provide the TSO number/title so I can read it?  I searched and didn't find anything about display standardization (but found many TSOs that are related to ADS-B traffic).  Here seems to be the most relevant FAA document on traffic alerts:
- https://www.faa.gov/nextgen/programs/adsb/pilot/atas
Note it does indicated target direction on the "dot" (in their case, a delta wing style triangle).

vancinad, 2020-07-06:
There are numerous illustrations of flight instruments using this symbology in the Instrument Flying Handbook (my copy is 8083-15B), chapter 5. 

Also reference AIM 4-5-2 where the ATC equivalent "Track velocity and direction vector line (projected ahead of target)" is depicted.
It is really standardized and while I agree it's not 100% intuitively obvious, once you know, you know.

 
 

https://flightaware.com/live/

https://www.aviationweather.gov/windtemp/plot
https://www.windy.com/pois?43.779,-82.617,6,i:pressure

JohnW4SBA, 2020-07-06:
> a key, which we don't currently have available in-app.

Documentation is sorely lagging development for Avare.  Maybe we could 
try to assemble a team to take it on at some point.  Videos also seem to 
be quite popular for a large segment of users and those too are mostly 
outdated and inadequate, tho far more work to create.  In the end 
though, our user base seems to include many people who make little or no 
effort to study all that's available, rather than "learn by doing" and 
asking questions on the Forum that demonstrate their approach.  I can 
make this claim, having not read the whole help file in a few years. ;)

JohnW4SBA, 2020-07-06:
> the "dot" (in their case, a delta wing style triangle).

A line-width arrowhead instead of a dot, with our current line from 
there for speed would actually obscure less of the chart while also 
providing direction at a glance.  A fair bit of code to orient the < 
arrowhead though?

apps4av, 2020-07-06:
Ok I will add a arrow head inside the circle.

gustafson, 2020-07-06:
Thanks Zubair,

vancinad,
I scanned the images in 8083-15B and see two as you described (figure 5-53 and 5-54 which essentially repeats 5-53).  However, also one figure 5-49 that has delta wings in addition to the line.  The later release of 8083-16B seems to have replaced all those images and the only image I found (in the corner of figure 5-6 in a non-exhaustive search) had triangles but no line.

And the AIM section you referenced was in regards to ATC screens rather than a panel instrument. Those likely hadn't been updated in I'm guessing 40 years and came with lots of specific training.  So I respectfully disagree that this is strong evidence of standardization for panel displays.

Of course, that doesn't mean that you are wrong.  :) I hope you feel Zubair has struck a reasonable compromise?  Thanks for the thoughtful debate.

vancinad, 2020-07-06:
Thanks Mr. G. My point was simply that to indicate velocity and direction with a line extending from the symbol is common both on the ground and in the air.
As for the proposed solution: If Zubair can easily put an arrow in the circle, that's great. If using the standard delta shapes is possible that seems like it would be even better. (Better as in "more conformant to the ADS-B standard symbology." If it costs a lot of processor power or memory or screen real estate then maybe it's not better, on balance.)
I'm sure something good will come. Thanks!

apps4av, 2020-07-06:
I looked in the code. The line symbology comes from Garmin GPS. It
indicates the ground track of the target in 1 minute. I think that's very
useful to check collision.
I also looked into changing circles to delta wings. It's not that simple
(memory use) and that's probably why I used circles.

saabnut, 2020-07-07:
I would not say flightaware has a more intuitive display. I agree regarding
directional ambiguity. I would love to see the age of the "ping" somehow
displayed (brightness?) and confidence (updating late or packet loss). In
theory we are all on on same clock, with 1s data intervals, yet airplanes
are not where expected at times. Is this due to processing of the data
packet or something else. (using stratux over wifi with both 978 and 1090,
and gps input.) I believe the triangles are from an early ARINC spec, like
735. Where closing speed is greater than 100, fast flash the icon.  I like
this display:
[image: image.png]

apps4av, 2020-07-11:
Added arrows over rectangles

closed 2020-07-11 by apps4av