apps4av/avare #377: Allow rotation for plates

n251ea, 2020-07-13:
some plates, specifically STARs and SIDs are printed in landscape instead of portrait  mode it would be useful to allow a 90 degree rotation even though the plate won't fit on the screen.

apps4av, 2020-07-22:
That will be scary in IMC.

apps4av, 2020-07-22:
By the way this is doable in Map screen. Plates rotate in Plates layer. Closing.

n251ea, 2020-07-22:
Either I don't understand how to rotate the plate or you don't understand what I'm looking for.Take for example the mmu 7 departure (attached), page 1 is a diagram that should be viewed in landscape mode, I am looking for the ability to rotate this image to view it.JeffSent from my cell phone

gustafson, 2020-07-22:
if I understand what Jeff is requesting, I wonder if this should be done instead in the data generation phase rather than in app. Recently I've decided to lock my tablet and landscape mode (

Reopened #377<https://github.com/apps4av/avare/issues/377>.

n251ea, 2020-08-26:
I think that makes perfect sense to do the rotation in the generation phase.  Only disadvantage is for the person that rotates their tablet to landscape mode, the page is still wrong by 90 degrees.