apps4av/avare #381: Destination weather without scrolling

gustafson, 2020-08-05:
A request from the forums.

Develop a method which displays destination weather without scrolling + long press.

shosk, 2020-08-06:
It would be very handy, when in route, to find the destination weather without scrolling through the charts, and with a minimum of screen changes. As I am currently using Avare, if I want to see destination weather I scroll up a couple hundred miles and then reverse pinch-to-zoom on the airport, then long press the airport button and hope I've got it centered exactly and then I can finally check destination weather. It would be nice if there was some way to tap on something on the home screen and see what the latest METAR.

apps4av, 2020-08-06:
How about using Plan tab weather briefing?

shosk, 2020-08-23:
It would be good if the destination weather page were quickly available from the home map screen. How about if we were to double tap (or triple tap) the green button? That could be a very useful function.

Thanks.

n251ea, 2020-08-26:
Here's a thought, from the "Find" page, there are already two selection modes, one is a quick press that sets your destination to the chosen waypoint, the other option is a long press on the waypoint and you are given the options to add it to your plan, delete it (not sure from what, the plan?) and bring up the plate, I'd propose two more options

1) bring up the Chart supplement for that waypoint - Including Navigational aids such as VORs and remaining NDBs
2) if there is WX for that waypoint  then display a box for WX and allow the WX to be displayed.