apps4av/avare #398: ADS-B WX not expiring

n251ea, 2020-09-07:
Performed two flights, almost 24hrs apart, when I opened Avare the second day I found it had retained all the METARs, TAFs, RADAR etc from the previous day.  

This data is stale, plus ADS-B refreshes this data often (multiple times per hour), it shouldn't be kept for any length of time as it can easily cause somebody to see the WX at a location which is incorrect if the FIS-B data is not being updated as expected.

JohnW4SBA, 2020-09-08:
> ADS-B refreshes this data often

Do you have "Use ADSB Weather" checked?

If so I don't think it matters what you have for Data Expiry, but maybe 
if there's no new ADSB acquired it doesn't change?

n251ea, 2020-09-08:
Yes "use ADS-B weather" is checked.  If there is no update the received data doesn't expire or at least stays available on the tablet for an extended amount of time. (see attached screenshots).  

Data expiration could be the issue as it is set to 1440.  The reason I have the data expiration set so long because many TFRs are not transmitted via FIS-B and/or Avare doesn't display received TFRs on the screen, the long expiration time ensures that TFRs don't expire from the display while they are still active.  I have found the presidential TFRs for whatever reason are not transmitted via FIS-B.

If new data is received via FIS-B then the data is updated, the concern comes with the software displaying stale WX data.

Possible display suggestions -
1) TFR data downloaded via the Internet should have its own expiration timer, maybe individual expiration timers?
2) Somehow mark displayed WX data that it hasn't been refreshed in a certain length of time, maybe outline the text portion of the WX with a red box?
3) RADAR images are only valid for a  short period of time, see #1? 
 
[image]
[image]
[image]