apps4av/avare #424: Avare crashes when set to sensor orientation

n251ea, 2021-02-18:
Avare 10.0.2 crashes when the orientation is set to Sensor and the tablet is rotated.

This happens on my Samsung tablets running Android 7.0, 7.1.1 and 9, it does not happen on my tablet running Android 10

n251ea, 2021-02-20:
I have been able to cause Avare to crash even on Android 10 by rotating the tablet, in addition if Avare is set in the application to portrait mode and you open Avare while the tablet is physically in a landscape Avare will open and immediately close

apps4av, 2021-02-20:
Can you get any crash logs? adb logcat command will do.
Thanks

n251ea, 2021-02-20:
[avarecrash.out.gz](https://github.com/apps4av/avare/files/6015536/avarecrash.out.gz)
See the attached file, I started "adb logcat" held the tablet in landscape mode, opened Avare, Avare quickly flashed on the screen showed the "tips" popup rotated to portrait mode and then exited.  I then stopped adb about 5 seconds later.  

Note Avare is set in its configuration to portrait mode.

apps4av, 2021-02-20:
Your phone is generating a message every 3 ms. There are just too many
messages to parse with too many app signatures. Maybe you need to find why
the phone is generating so many messages. It could be decreasing phone
performance.